cameroncros/OctoPrint-DiscordRemote #7: created a bot

jon012198, 2018-06-06:
it dose not post or listen to the channel i did give it permissions

jon012198, 2018-06-06:
i found out i did not have dev mode on

cameroncros, 2018-06-06:
The instructions to set it up could be more clear. I will try write something up at some point. Will keep this open to track the issue

jon012198, 2018-06-06:
i did get it to work https://puu.sh/AAq7v/e3907a2ea1.png

jon012198, 2018-06-06:
i used this https://github.com/Chikachi/DiscordIntegration/wiki/How-to-get-a-token-and-channel-ID-for-Discord and this https://github.com/reactiflux/discord-irc/wiki/Creating-a-discord-bot-&-getting-a-token

jon012198, 2018-06-08:
how do you make a bot stay online

cameroncros, 2018-06-08:
Fixed doco here: 70a872a0c0205487963df8c17a1c0fed67a746da

As for the bot staying online, not sure. I may have missed something when i created the discord bot (AFAIK there are no python 2 discord bots apart from this one). I will look into it more. See #10 as it is probably related.

closed 2018-06-08 by cameroncros

jon012198, 2018-06-08:
i did not program my bot that is why i need to do that

jon012198, 2018-06-08:
> Sorry, I don't know what you mean? Is your bot connecting and then later
> disconnecting? Or is it not connecting at all?

Bot was there then it went offline