cameroncros/OctoPrint-DiscordRemote #10: Discord bot connecting 1000 times in 24 hours

littleferris, 2018-06-07:
I got this today. Not sure if its something that needs to be done with the code or if its just a fluke.

[image]

cameroncros, 2018-06-08:
Honestly, dont know, but suspect its the code is wrong. I will investigate further. I have seen this once before, but I had a terrible internet connection, so I assumed it was that.

jon012198, 2018-06-09:
i just got the same message

cameroncros, 2018-06-12:
I have worked out how to get the bot to remain up/reconnect after a network drop. Still haven't managed to replicate the 1000 reconnects though. Will clean up code and commit and merge, and maybe it will work?

lewnewby, 2018-06-12:
I have had this occur two days running, the print completes and then I get the email from Discord 3 hours later. 

This does not appear to be a network drop, the timing is too consistent and I don't believe my network is that fragile.

Looking at the log discordremote  appears to get a connection close after 55 secs and then reopens the connection immediately, this equates to 1000 connections in just over 15 hours. 

Once discord changes the token because of the the "More than 1000 connection in a short time error" discordremote goes to attempting every second. I have a log file that shows this behavior though it only has INFO messages.

cameroncros, 2018-06-12:
@lewnewby Can you please attach the log or send it to me somehow?

lewnewby, 2018-06-13:
Here you go
[octoprint.log.zip](https://github.com/cameroncros/OctoPrint-DiscordRemote/files/2096291/octoprint.log.zip)

cameroncros, 2018-06-13:
Thanks @lewnewby I'll look into that more. I honestly have no idea how your in that state.

I have updated the discord code, and it now successfully resumes a broken connection (Also fair bit more logging). Please update and see if you still have the issue.

jon012198, 2018-06-13:
what ever you add since 2.2 broke it and it still dose it and i looked at your code look like when it get closed it open a new message to send  i think  if can make it send it one every 24 hours would be better
[octoprint2-3.log](https://github.com/cameroncros/OctoPrint-DiscordRemote/files/2098142/octoprint2-3.log)

cameroncros, 2018-06-13:
@jon012198 and @lewnewby Both of your logs start with "--- Log roll over detected ---", which means I cant see what originally happened to cause the issue. Can you try restarting octoprint, and getting the log from immediately after that?

cameroncros, 2018-06-13:
[samplelog.log](https://github.com/cameroncros/OctoPrint-DiscordRemote/files/2098657/samplelog.log)
Something like the above? Note the "Starting Octoprint 1.3.8" line.

lewnewby, 2018-06-13:
Here you go, The restart is at line 231 in octoprint.log.2018-06-10
[Octoprint-logs.zip](https://github.com/cameroncros/OctoPrint-DiscordRemote/files/2098756/Octoprint-logs.zip)

lewnewby, 2018-06-13:
Yep confirmed, just over 15 hours and Discord throws the more than 1000 connection errors

cameroncros, 2018-06-13:
Yup. Thanks for that log, very interesting:
`2018-06-10 19:50:51,535 - octoprint.plugins.discordremote - ERROR - <Response [404]>: {"code": 0, "message": "404: Not Found"} - {'Expect-CT': 'max-age=604800, report-uri="https://report-uri.cloudflare.com/cdn-cgi/beacon/expect-ct"', 'Content-Length': '40', 'Via': '1.1 google', 'Set-Cookie': '__cfduid=d3eb932dd0b6b5bd3dcf8f15f5f39f51f1528660251; expires=Mon, 10-Jun-19 19:50:51 GMT; path=/; domain=.discordapp.com; HttpOnly', 'Strict-Transport-Security': 'max-age=31536000; includeSubDomains', 'Server': 'cloudflare', 'Connection': 'keep-alive', 'Date': 'Sun, 10 Jun 2018 19:50:51 GMT', 'CF-RAY': '428e5e8bce48831b-ATL', 'Alt-Svc': 'clear', 'Content-Type': 'application/json'}`

If you go to https://discordapp.com/api/gateway in a browser, does it work?
Can you do a wget or curl from your octoprint server? eg: wget https://discordapp.com/api/gateway

lewnewby, 2018-06-13:
Yes to the first question and here is the output of the second

pi@octopi:~/.octoprint/logs $ wget https://discordapp.com/api/gateway
--2018-06-13 15:36:25--  https://discordapp.com/api/gateway
Resolving discordapp.com (discordapp.com)... 104.16.58.5, 104.16.59.5
Connecting to discordapp.com (discordapp.com)|104.16.58.5|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 35 [application/json]
Saving to: ‘gateway’

gateway             100%[=====================>]      35  --.-KB/s   in 0s

2018-06-13 15:36:25 (8.11 MB/s) - ‘gateway’ saved [35/35]

jon012198, 2018-06-13:
i get the same stuff

lewnewby, 2018-06-13:
Doing some research, at https://discordapp.com/developers/docs/topics/gateway and found this Note:

> Clients are limited to 1000 IDENTIFY calls to the websocket in a 24-hour period. This limit is global and across all shards, but does not include RESUME calls. Upon hitting this limit, all active sessions for the bot will be terminated, the bot's token will be reset, and the owner will receive an email notification. It's up to the owner to update their application with the new token.

It appears to me that the bot isn't processing the READY event from the gateway and thus doesn't have a session_ID stored to then be checked for in the RESUME code (discord.py:105) when the gateway disconnects.

cameroncros, 2018-06-13:
The new bot is doing the resume code, but not on the ready event, maybe I should put it there though. Im still curious about the 404 error.

jon012198, 2018-06-14:
looks like it is fixed  here is the log so you can check and i did put you plugin in debug to help

[octoprint2.4.log](https://github.com/cameroncros/OctoPrint-DiscordRemote/files/2102478/octoprint2.4.log)

cameroncros, 2018-06-14:
Huh, weird, i just got failures on my end :( I think there is more work to fix everything, so i'll keep at it.

jon012198, 2018-06-14:
i did restart my pi completely so you could have something cached @cameroncros

cameroncros, 2018-06-14:
Don't upload the logs to github anymore, with the extended debugging they are printing stuff like your channel id and users. I don't know if they are abuseable pieces of info, but to be safe, don't upload them. (Perhaps email them might be safer?)

On a related note, I am rearchitecting the threading, and I think I have got it a bit nicer. Will commit and release later, but want to test it thoroughly. I think i can now test the restarting 1000 times thing now, so I will try do that as well.

jon012198, 2018-06-18:
did a restart of you pi fix the discord spam or not sure

cameroncros, 2018-06-18:
Still not sure. I think i have the reconnection code working properly, but I still cant reliably reproduce the reconnect spam. In a few days time I'll probably release a new version with all the changes I have made, and i guess see how that goes?

jon012198, 2018-06-18:
ok  lets hops it works out and dose not brake it more i tend to brake
people thing a lot / find bugs

cameroncros, 2018-06-19:
v2.5 is out, has minor improvements to reliability. If I dont hear any new complaints in the next week or two, I might close this ticket? I do want to get it fixed, but without any idea whats going on, its really hard.

monkphin, 2018-08-14:
Just had the same email. Granted I did a LOT of quick prints a few days ago, but not enough to generate 1K+ messages. 

Will pull the logs later today when I’m in front of a PC.

cameroncros, 2018-08-14:
@monkphin Can you tell me which version of the DiscordRemote plugin you currently have? AFAIK, no one has had this issue in a long time, so I am surprised to hear its happening again. Any logging information you have would be very useful.

monkphin, 2018-08-15:
@cameroncros currently running 2.16.0 - which is the latest version shown to me on Octorint currently. I've attached the last fdew days worth of log outputs. The last time I had anything communicated into Discord by DiscordRemote was the 12th, with the email about exceeding the 1000 connection limit being sent at 17:50 (BST) yesterday. No settings had been changed on Discord Remote between the 12th and by the time the notification email came through. 

[octoprint 15-08-2018.log.log](https://github.com/cameroncros/OctoPrint-DiscordRemote/files/2290096/octoprint.15-08-2018.log.log)
[octoprint 13-08-2018.log](https://github.com/cameroncros/OctoPrint-DiscordRemote/files/2290100/octoprint.13-08-2018.log)
[octoprint 12-08-2018.log](https://github.com/cameroncros/OctoPrint-DiscordRemote/files/2290101/octoprint.12-08-2018.log)

cameroncros, 2018-08-16:
There are no more stability changes added since 2.16, but there are a few more releases since then. Looking at your logs, I have the following guess:

- The RESUME message is following an IDENTIFY message. AFAIK, the 1000 connection limit is for IDENTIFY messages, not RESUME. I think the fix here is to send a RESUME instead of IDENTIFY, if we have previously been connected.

jon012198, 2018-08-18:
same thing as they got  same log

cameroncros, 2018-08-19:
Interesting that all 3 of you got it at the same time. I am testing the disconnect/reconnect code currently, but it is taking a while (1 minute per reset * 1000 is a long time :( )

cameroncros, 2018-08-19:
Test failed, still not sure why it is happening...

latchspring, 2018-08-20:
watching this thread too as I'm having the same issue.

cameroncros, 2018-08-21:
Im getting pretty confused. I have been following the discord.py codebase, and it does resuming the same way as I have done it. So not sure what the difference is. Im gonna have a play with the discord.py library and see if it does something I am not doing. Ideally, when octoprint eventually gets to python 3, I will do a rewrite of this plugin to use discord.py

cameroncros, 2018-08-21:
https://github.com/b1naryth1ef/disco <- just found this, will try intergrate this instead of the halfbaked discord bot i wrote. My one makes me cry anyway.

cameroncros, 2018-08-21:
Okay, I think I have a solution to the sailing to resume issues, will test and push an update when I'm happy with it. Unfortunately, still no idea why you all keep getting thousands of disconnects in 24 hours, which would be good to solve as well

Manichee, 2018-08-21:
May or may not have a bearing on this, but I have not had that issue since I restarted octoprint right after I posted here. I only had the issue on the very first startup after installing the latest version. Granted I haven't really done interactions with bot since then. This is the current log:
2018-08-21 15:59:05,390 - octoprint.plugins.discordremote - INFO - Heartbeat: {"d": 3, "op": 1}
2018-08-21 15:59:05,433 - octoprint.plugins.discordremote - INFO - Received HEARTBEAT_ACK message

cameroncros, 2018-08-22:
Okay, for some reason, the heartbeat thread is exiting. It shouldn't logically happen, as far as I can tell (if someone spots a flaw in my code related to this, would be apprecieated). I have added more debugging messages and hopefully made it more resilient.
I have also fixed up the RESUME handling, so that should also prevent the error causing your bot tokens getting expired.

Kirby207, 2018-08-23:
> Sorry for the delay fixing all of this.

Bud, you've got nothing to apologize for. this bot is awesome and aside from people putting up logs, you've been on your own. Keep on keepin' on.

And for what it might or might not be worth - I've literally _never_ had this issue occur to me, been using the bot since Jun 14th or so. My only setup difference is OctoPrint isn't running on a RPi, it's on a spare laptop running Ubuntu server 16.04 that's dedicated to OctoPrint.

cameroncros, 2018-08-23:
I suspect it's more likely related to people's less reliable internet connections, I have only seen it happen once to me, and I have a very reliable internet.
If I'm right about the reason it happens, then it would be rare to happen anyway.

lewnewby, 2018-08-23:
I haven't looked at the code recently but having gone through the Discord API's and working with a number of API's there can be no expectation of stable network connectivity. I for one have an extremely stable connection but that does not mean that the routes between me and another location will remain stable. 

While writing this I went and had a look at the code in the dev branch and what you have done in the handle_hello method I feel is exactly what was needed and what I was talking about on June 13.

cameroncros, 2018-08-23:
Yeah, of course discord apis don't require a stable network connection, but my poor implementation meant that an unstable connection could cause the issue, and a stable connection would make it. I had the resume code for a while, but it was also broken, because I was closing the session with the ws.close(). (Had some help from another discord dev for that one)

lewnewby, 2018-08-24:
Looks good Cameron and I will give it a shot once more or at least when I get my printer up and running again (tinkering is a vice LOL )

cameroncros, 2018-09-15:
I'm gonna once again assume that it has finally been fixed, so I'll close this issue. Thanks for all the logs and patience.

cameroncros, 2018-12-24:
Can you post some logs @ducky12432?

cameroncros, 2019-01-04:
@ducky12432 If you have no logs available, I cannot debug this issue. I'm going to re-close this ticket.

ducky12432, 2019-01-04:
Sorry I have been away for a couple of weeks and did not see this.  I will post some logs once I have gotten home

ducky12432, 2019-01-12:
I am back now and have reconnected the bot and will attempt to recreate the issue

ducky12432, 2019-01-23:
Yea I finally tracked the issue down to plugins are not updating even though they say they are.  I think it may finally be time for a reinstall and reconfigure.

berkeoznalbant, 2021-01-08:
Getting this error for a week now. Tried reinstalling the plugin but nothing changed. Here is the [log](https://github.com/cameroncros/OctoPrint-DiscordRemote/files/5789385/octoprint.7.log)

cameroncros, 2021-01-09:
@berkeoznalbant Thanks for the report, I am guessing the error handling code is still not perfect. I will look into it when I get a chance.

berkeoznalbant, 2021-01-09:
I think after regenerating a token for the bot it is fixed. I'll let you know if it gives error again. BTW, bot was not responding to my discord commands but was updating in discord during prints. Now, it is working fine though.

cameroncros, 2021-01-09:
You'll have the issue again soon, it's because the bot is spamming it's
connections, so discord blocks the token. I will look into a fix asap

closed 2021-10-17 by cameroncros