cameroncros/OctoPrint-DiscordRemote #14: [request] message test

jon012198, 2018-06-16:
a button to ping the discord bot so i dont have to remove one charter to reload the  bots connection

jon012198, 2018-06-16:
> Just to be clear, you want a button on the web interface that will send a
> test message to discord?
> While im at it, I might put an indicator to see if the bot is connected
> and listening.

Yup

cameroncros, 2018-06-18:
I have added an indicator that shows the status of the discord bot. Still trying to work out the best way to send a test message from the web UI.

cameroncros, 2018-06-19:
53382da7af005bf79e4dd081cd68932a65a627c3 Show status indicator in web interface
81fbd577504c709b648de21c2ac453662c115659 Send message from web interface (default status)

jon012198, 2018-06-19:
formatting is off only when you use the button in octoprint but if you send a message in discord it is fine 
[image]
[image]

cameroncros, 2018-06-19:
Whoops, only tested with status, and didnt look too carefully at alignment, will fix for next release.

cameroncros, 2018-06-21:
Fixed the messages, and have made the tables a bit thinner, should print much nicer now.

closed 2018-06-21 by cameroncros