cameroncros/OctoPrint-DiscordRemote #15: mobile

jon012198, 2018-06-22:
formatting on mobile is bad  the status one is fine on mobile https://cdn.discordapp.com/attachments/423501252618485770/459528563129843722/image.png

cameroncros, 2018-06-22:
But my beautiful ascii tables :(

In seriousness, probably would be good to get something more like that. Im not in a huge hurry to implement it though, so if you are about to do it, and it is something you want done, feel free to go for it. Otherwise I might get around to in in the future. If you want help with setting it up, let me know.

cameroncros, 2018-06-24:
embeds have pretty bad limitations, as described by the link above. Not sure they are suitable. The markdown option may be worth exploring

cameroncros, 2018-07-09:
markdown is pretty limited as well, might have to do more research

cameroncros, 2018-07-19:
Markdown inside code blocks might be a way forward, I added the initial steps by adding a way to format the messages as markdown code blocks, will try add a table formatter that generates nice markdown tables, and see if that looks any nicer

cameroncros, 2018-07-25:
I have created a embed based solution, but want to test it a bit more thoroughly. If your interested, install: https://github.com/cameroncros/OctoPrint-DiscordRemote/archive/DR-15.zip via the plugin manager.

jon012198, 2018-07-25:
> cameroncros/OctoPrint-DiscordRemote/archive/DR-15.zip via the plugin

it looks good to me anything you want me to test  all of the commands work
right

cameroncros, 2018-07-26:
If it is good for you, I will do a bit more cleanup and then release it for everyone
It is possible to send snapshots inside embeds, which i want to try as well, but might save that for later.

cameroncros, 2018-07-27:
Added snapshots to the embeds. Will need to test with a real printer, but once that is done, I will merge and complete it. #31

cameroncros, 2018-07-28:
Merged, if CI passes, will release

closed 2018-07-28 by cameroncros