cameroncros/OctoPrint-DiscordRemote #16: Bot responds to all messages typed in channel regardless of prefix

lewnewby, 2018-06-27:
I don't know if I missed something during setup but the bot has started responding to anything a person says in the channel with the help message.

Attached is a screenshot.
[image]

cameroncros, 2018-06-28:
Yup, that was by design. If the bot doesn't understand what you are saying, it will print the help for you. I suggest giving the bot its own channel.

cameroncros, 2018-06-28:
Yeah, okay, I will look into that. Bear in mind, that allowing others to chat in the discord channel does allow them to control your printer. (Which could be _very_ unsafe). I will change it so that / anything returns the help, along with "help" (possibly caseless)

tdub415, 2018-06-28:
I've already got it locked down where the bot will only respond when I post by changing line 213 of discord.py from `        if "bot" in data['author']:` to `        if data['author']['id'] != "#######":` the ###### being my unique discord ID number.

lewnewby, 2018-06-28:
Ok well you answered my question. I have disabled the bot and plugin. Perhaps in the future I'll go back to it.

cameroncros, 2018-06-29:
@tdub415 See #18. I didn't originally want to do this, because doing permissions right is hard, and I don't want to be responsible for messing it up. But seeing as its something at least 2 of you seem to want, I will work on implementing it.

tdub415, 2018-06-29:
Neat, the method I'm using is what a buddy came up with, and it works well so far.  Not sure how easy it would be to add a fill in box similar to the channel token but in the config to put a user ID in.

cameroncros, 2018-06-29:
That was exactly the plan. Comma seperated to add multiple users.

cameroncros, 2018-07-02:
Bot now only listens for /* commands, and also "help" (case insensitive)

closed 2018-07-02 by cameroncros