cameroncros/OctoPrint-DiscordRemote #17: Plugin using vast amounts of upstream bandwidth (10+ Mbps!) talking to Cloudflare (report-uri?) (when unconfigured?)

aziraphale, 2018-06-28:
Hi,

I installed this plugin but didn't get around to configuring it yet, so it's been sat idle for a few weeks, I guess. Today I checked out the usage graphs for my ISP and noticed that since around midday (BST) on Saturday, we've been using a huge amount of upstream bandwidth - on the order of 50 GiB/day.

I tracked it down to my Octoprint Pi, down to Octoprint itself, and after running a packet sniffer on my router, which returned packets referencing `discordapp.com`, I traced it to this plugin. Restarting Octoprint saw the data spam continue, albeit at a much lower rate (judging by the graphs for my ISP, the data stream gradually increases over the course of 60-90 minutes). When I disabled this plugin and restarted Octoprint, the data spam ceased entirely, with just a couple of kbps background traffic remaining.

Graph from Saturday, the red line is our uploads, showing around 5 Mbps, but note that we have two bonded VDSL lines, so seeing 5 Mbps on one line means that there's also 5 Mbps on the other line for a total of 10 Mbps:
![A&A ISP CQM graph for Saturday 23rd June showing upstream bandwidth usage being almost non-existent until around 12:30pm at which point it gradually climbs, over the course of a few hours, up to about 5 Mbps](https://i.imgur.com/aN0w7z7.png "Saturday's CQM Graph")

Graph from Sunday (Monday, Tuesday and Wednesday look much the same):
[image]

The traffic was all going to a Cloudflare IP address, `104.16.59.5`.

Upon checking my Octoprint logs folder I found that my `octoprint.log` file was 36 MiB, but previous days' logs were more than 200 MiB.

The most recent `octoprint.log` almost entirely consists of these lines repeated forever:

```
2018-06-27 12:24:02,508 - octoprint.plugins.discordremote - INFO - Message queued 2
2018-06-27 12:24:04,162 - octoprint.plugins.discordremote - ERROR - <Response [404]>: {"code": 0, "message": "404: Not Found"} - {'Expect-CT': 'max-age=604800, report-uri="https://report-uri.cloudflare.com/cdn-cgi/beacon/expect-ct"', 'Content-Length': '40', 'Via': '1.1 google', 'Set-Cookie': '__cfduid=dc0ac58b2ced576345210d7d85df498be1530098643; expires=Thu, 27-Jun-19 11:24:03 GMT; path=/; domain=.discordapp.com; HttpOnly', 'Strict-Transport-Security': 'max-age=31536000; includeSubDomains', 'Server': 'cloudflare', 'Connection': 'keep-alive', 'Date': 'Wed, 27 Jun 2018 11:24:04 GMT', 'CF-RAY': '43178b876ae36b55-LHR', 'Alt-Svc': 'clear', 'Content-Type': 'application/json'}
```

However the older log files (Saturday and earlier) also include some HTML alongside the `report-uri` line:

```
<!--[if lt IE 7]> <html class="no-js ie6 oldie" lang="en-US"> <![endif]-->
<!--[if IE 7]>    <html class="no-js ie7 oldie" lang="en-US"> <![endif]-->
<!--[if IE 8]>    <html class="no-js ie8 oldie" lang="en-US"> <![endif]-->
<!--[if gt IE 8]><!--> <html class="no-js" lang="en-US"> <!--<![endif]-->
<head>

<title>discordapp.com | 502: Bad gateway</title>
<meta charset="UTF-8" />
<meta http-equiv="Content-Type" content="text/html; charset=UTF-8" />
<meta http-equiv="X-UA-Compatible" content="IE=Edge,chrome=1" />
<meta name="robots" content="noindex, nofollow" />
<meta name="viewport" content="width=device-width,initial-scale=1,maximum-scale=1" />
<link rel="stylesheet" id="cf_styles-css" href="/cdn-cgi/styles/cf.errors.css" type="text/css" media="screen,projection" />
<!--[if lt IE 9]><link rel="stylesheet" id='cf_styles-ie-css' href="/cdn-cgi/styles/cf.errors.ie.css" type="text/css" media="screen,projection" /><![endif]-->
<style type="text/css">body{margin:0;padding:0}</style>

</head>
<body>
<div id="cf-wrapper">

    <div id="cf-error-details" class="cf-error-details-wrapper">
        <div class="cf-wrapper cf-error-overview">
            <h1>

              <span class="cf-error-type">Error</span>
              <span class="cf-error-code">502</span>
              <small class="heading-ray-id">Ray ID: 42e1069a3c146ba3 &bull; 2018-06-20 20:36:07 UTC</small>
            </h1>
            <h2 class="cf-subheadline">Bad gateway</h2>
        </div><!-- /.error-overview -->

        <div class="cf-section cf-highlight cf-status-display">
            <div class="cf-wrapper">
                <div class="cf-columns cols-3">

<div id="cf-browser-status" class="cf-column cf-status-item cf-browser-status ">
  <div class="cf-icon-error-container">
    <i class="cf-icon cf-icon-browser"></i>
    <i class="cf-icon-status cf-icon-ok"></i>
  </div>
  <span class="cf-status-desc">You</span>
  <h3 class="cf-status-name">Browser</h3>
  <span class="cf-status-label">Working</span>
</div>

<div id="cf-cloudflare-status" class="cf-column cf-status-item cf-cloudflare-status ">
  <div class="cf-icon-error-container">
    <i class="cf-icon cf-icon-cloud"></i>
    <i class="cf-icon-status cf-icon-ok"></i>
  </div>
  <span class="cf-status-desc">London</span>
  <h3 class="cf-status-name">Cloudflare</h3>
  <span class="cf-status-label">Working</span>
</div>

<div id="cf-host-status" class="cf-column cf-status-item cf-host-status cf-error-source">
  <div class="cf-icon-error-container">
    <i class="cf-icon cf-icon-server"></i>
    <i class="cf-icon-status cf-icon-error"></i>
  </div>
  <span class="cf-status-desc">discordapp.com</span>
  <h3 class="cf-status-name">Host</h3>
  <span class="cf-status-label">Error</span>
</div>

                </div>

            </div>
        </div><!-- /.status-display -->

        <div class="cf-section cf-wrapper">
            <div class="cf-columns two">
                <div class="cf-column">
                    <h2>What happened?</h2>
                    <p>The web server reported a bad gateway error.</p>
                </div>

                <div class="cf-column">
                    <h2>What can I do?</h2>
                    <p>Please try again in a few minutes.</p>
                </div>
            </div>

        </div><!-- /.section -->

        <div class="cf-error-footer cf-wrapper">
  <p>
    <span class="cf-footer-item">Cloudflare Ray ID: <strong>42e1069a3c146ba3</strong></span>
    <span class="cf-footer-separator">&bull;</span>
    <span class="cf-footer-item"><span>Your IP</span>: 90.155.91.100</span>
    <span class="cf-footer-separator">&bull;</span>
    <span class="cf-footer-item"><span>Performance &amp; security by</span> <a href="https://www.cloudflare.com/5xx-error-landing?utm_source=error_footer" id="brand_link" target="_blank">Cloudflare</a></span>

  </p>
</div><!-- /.error-footer -->

    </div><!-- /#cf-error-details -->
</div><!-- /#cf-wrapper -->
</body>
</html>
 - {'Expect-CT': 'max-age=604800, report-uri="https://report-uri.cloudflare.com/cdn-cgi/beacon/expect-ct"', 'Content-Length': '4093', 'Set-Cookie': '__cfduid=d86c5c4d1c8933a36c592d26952c5991e1529526967; expires=Thu, 20-Jun-19 20:36:07 GMT; path=/; domain=.discordapp.com; HttpOnly, cf_use_ob=0; expires=Wed, 20-Jun-18 20:36:37 GMT; path=/', 'Expires': 'Thu, 01 Jan 1970 00:00:01 GMT', 'Server': 'cloudflare', 'Connection': 'keep-alive', 'Pragma': 'no-cache', 'Cache-Control': 'no-store, no-cache, must-revalidate, post-check=0, pre-check=0', 'Date': 'Wed, 20 Jun 2018 20:36:07 GMT', 'X-Frame-Options': 'SAMEORIGIN', 'Content-Type': 'text/html; charset=UTF-8', 'CF-RAY': '42e1069a3c146ba3-LHR'}
```

My guess is that this plugin was attempting to do something, accidentally ended up being very spammy to someone (possibly the Report-URI service), and then that person blocked my Pi's IP address at Saturday lunchtime (the guy who runs Report-URI lives in the UK, so that's a perfectly reasonable time for him to be dealing with things like that), and then this plugin wasn't expecting the Report URI requests to 404 and just started spamming those requests, somehow doing so at increasing frequency - every 1-2 seconds judging by this `octoprint.log`!

But I suspect that the main issue will be that there is no logic to keep the plugin in "idle"/standby mode if it's not configured :)

I'm happy to share my log files privately, along with the packet dumps I collected, if that helps.

And hopefully this hasn't resulted in my entire IP address range being blocked by Report-URI, as that was a service I was planning to use at some point. It won't matter too much if only the Raspberry Pi's IP address was blocked, but if the block extends to my whole IP range I won't be happy =/

Thanks :)

cameroncros, 2018-06-28:
This is definitely a problem worth investigating. I will get onto this immediately.

cameroncros, 2018-07-02:
I have changed it so that after 50 errors in a row, the bot will shutdown. Im not completely happy with this alternative, but its something. I also made the bot not start if not properly configured, which should cut down on the misconfigured spamming.

closed 2018-07-02 by cameroncros

aziraphale, 2018-07-02:
Cool, that should sort it. Thanks! It looks like a great plugin, I just haven't had the free time to configure it and try it out yet! :)