cameroncros/OctoPrint-DiscordRemote #19: Bot always offline, Doesn't respond to commands.

xXRoboBoyXx, 2018-07-01:
I've setup the bot, it appears in my discord channel. I put the bot token and channel id into the plugin, but nothing happens. I've tried just about everything to get this plugin to work. I've triple checked that the bot token and channel id are correct.
[image]
[image]

cameroncros, 2018-07-02:
Can you email the full logs? Or attach them if your comfortable (I am not sure its 100% safe to do so, definitely get a new bot token afterwards)

xXRoboBoyXx, 2018-07-02:
[octoprint.log](https://github.com/cameroncros/OctoPrint-DiscordRemote/files/2153446/octoprint.log)
Here you go

cameroncros, 2018-07-02:
Thanks, im still lost unfortunately. Can you ping discordapp.com from the machine you are running the server from?

xXRoboBoyXx, 2018-07-02:
Sucessfully pinged discordapp.com

tdub415, 2018-07-03:
I've noticed recently that the bot part of plugin isn't consistently staying online as well.  The automatic messages during prints will still send when the bot decides to be offline, but it won't respond to commands.  Sometimes the bot will stay up for a few days after a reboot, sometimes only a few hours.

cameroncros, 2018-07-06:
@xXRoboBoyXx @tdub415 I have added a heap more logging to release 2.10. Can you both try it out and see if the logs show any more useful information?

xXRoboBoyXx, 2018-07-06:
[octoprint.log](https://github.com/cameroncros/OctoPrint-DiscordRemote/files/2170461/octoprint.log)
For some reason it's decided to work after i updated. I'll attach the log just in case.

tdub415, 2018-07-06:
So far so good, the bot hasn't gone offline since I updated to 2.10, but I haven't printed anything since updating.  I'll get something started printing this evening and keep an eye on it and post the log if it goes offline but still sends progress messages like it has been when it's gone offline.

xXRoboBoyXx, 2018-07-06:
I'm 100% sure that i'm using a bot token, and not a webhook. I do get the settings are correct message. The controller icon sends messages just fine. 
[image]
[image]

tdub415, 2018-07-06:
You're not using your discord snowflake ID, you can't just use your username. https://discordapp.com/developers/docs/reference#snowflakes

cameroncros, 2018-07-06:
Ah, you need to change that bot token now, and as fast as possible (anyone can use it now). The allowed users needs to be your user id, which is a number, similar to the channel id. (What @tdub415 said, thanks :) )

xXRoboBoyXx, 2018-07-06:
Oh, Everything's working now, thanks a bunch!

cameroncros, 2018-07-06:
Please renew your bot token. Anyone who has seen it can use the bot token to read messages on your channel, and do potentially bad things.

cameroncros, 2018-07-06:
Im gonna leave this open incase @tdub415 has any issues.

tdub415, 2018-07-10:
It still seems to be having issues staying online.  I'm not enough of a guru to really debug it much, but looking at the log, this appears to be 'where' it disconnected: `2018-07-10 13:50:57,847 - octoprint.plugins.discordremote - INFO - Heartbeat: {"d": 4614, "op": 1}
2018-07-10 13:51:38,892 - octoprint.plugins.discordremote - ERROR - Haven't received a heartbeat ACK in a while
2018-07-10 13:51:38,894 - octoprint.plugins.discordremote - INFO - Restart Triggered
2018-07-10 13:51:41,900 - octoprint.plugins.discordremote - INFO - Waiting for listener thread to join.
2018-07-10 13:52:19,939 - octoprint.plugins.discordremote - ERROR - Haven't received a heartbeat ACK in a while
2018-07-10 13:53:00,984 - octoprint.plugins.discordremote - ERROR - Haven't received a heartbeat ACK in a while
2018-07-10 13:53:42,029 - octoprint.plugins.discordremote - ERROR - Haven't received a heartbeat ACK in a while
2018-07-10 13:54:23,074 - octoprint.plugins.discordremote - ERROR - Haven't received a heartbeat ACK in a while
2018-07-10 13:55:04,121 - octoprint.plugins.discordremote - ERROR - Haven't received a heartbeat ACK in a while
2018-07-10 13:55:45,167 - octoprint.plugins.discordremote - ERROR - Haven't received a heartbeat ACK in a while
2018-07-10 13:56:26,213 - octoprint.plugins.discordremote - ERROR - Haven't received a heartbeat ACK in a while
2018-07-10 13:57:07,260 - octoprint.plugins.discordremote - ERROR - Haven't received a heartbeat ACK in a while
2018-07-10 13:57:48,306 - octoprint.plugins.discordremote - ERROR - Haven't received a heartbeat ACK in a while
2018-07-10 13:58:29,348 - octoprint.plugins.discordremote - ERROR - Haven't received a heartbeat ACK in a while
2018-07-10 13:59:10,394 - octoprint.plugins.discordremote - ERROR - Haven't received a heartbeat ACK in a while
2018-07-10 13:59:51,439 - octoprint.plugins.discordremote - ERROR - Haven't received a heartbeat ACK in a while
2018-07-10 14:00:32,480 - octoprint.plugins.discordremote - ERROR - Haven't received a heartbeat ACK in a while
2018-07-10 14:01:13,523 - octoprint.plugins.discordremote - ERROR - Haven't received a heartbeat ACK in a while
2018-07-10 14:01:54,567 - octoprint.plugins.discordremote - ERROR - Haven't received a heartbeat ACK in a while
2018-07-10 14:01:55,317 - octoprint.server.heartbeat - INFO - Server heartbeat <3
2018-07-10 14:02:35,612 - octoprint.plugins.discordremote - ERROR - Haven't received a heartbeat ACK in a while
2018-07-10 14:03:16,655 - octoprint.plugins.discordremote - ERROR - Haven't received a heartbeat ACK in a while
2018-07-10 14:03:57,702 - octoprint.plugins.discordremote - ERROR - Haven't received a heartbeat ACK in a while
2018-07-10 14:04:38,748 - octoprint.plugins.discordremote - ERROR - Haven't received a heartbeat ACK in a while
2018-07-10 14:05:19,796 - octoprint.plugins.discordremote - ERROR - Haven't received a heartbeat ACK in a while
2018-07-10 14:06:00,841 - octoprint.plugins.discordremote - ERROR - Haven't received a heartbeat ACK in a while
2018-07-10 14:06:41,887 - octoprint.plugins.discordremote - ERROR - Haven't received a heartbeat ACK in a while
2018-07-10 14:07:22,932 - octoprint.plugins.discordremote - ERROR - Haven't received a heartbeat ACK in a while
2018-07-10 14:08:03,976 - octoprint.plugins.discordremote - ERROR - Haven't received a heartbeat ACK in a while
2018-07-10 14:08:45,022 - octoprint.plugins.discordremote - ERROR - Haven't received a heartbeat ACK in a while
2018-07-10 14:09:26,065 - octoprint.plugins.discordremote - ERROR - Haven't received a heartbeat ACK in a while
2018-07-10 14:10:07,112 - octoprint.plugins.discordremote - ERROR - Haven't received a heartbeat ACK in a while
2018-07-10 14:10:48,158 - octoprint.plugins.discordremote - ERROR - Haven't received a heartbeat ACK in a while
2018-07-10 14:11:29,204 - octoprint.plugins.discordremote - ERROR - Haven't received a heartbeat ACK in a while
2018-07-10 14:12:10,250 - octoprint.plugins.discordremote - ERROR - Haven't received a heartbeat ACK in a while
2018-07-10 14:12:51,295 - octoprint.plugins.discordremote - ERROR - Haven't received a heartbeat ACK in a while
2018-07-10 14:13:32,339 - octoprint.plugins.discordremote - ERROR - Haven't received a heartbeat ACK in a while
2018-07-10 14:14:13,384 - octoprint.plugins.discordremote - ERROR - Haven't received a heartbeat ACK in a while
2018-07-10 14:14:54,428 - octoprint.plugins.discordremote - ERROR - Haven't received a heartbeat ACK in a while
2018-07-10 14:15:35,475 - octoprint.plugins.discordremote - ERROR - Haven't received a heartbeat ACK in a while
2018-07-10 14:16:16,521 - octoprint.plugins.discordremote - ERROR - Haven't received a heartbeat ACK in a while
2018-07-10 14:16:55,318 - octoprint.server.heartbeat - INFO - Server heartbeat <3`

cameroncros, 2018-07-11:
Thanks for that, based on the info there, it seems that the websocket thread is not cleaning up properly as expected. I dont really know how to best handle this, so I think abandoning/leaking the hung thread after a few seconds might be the best option. Will update and push a new release later this week.

cameroncros, 2018-07-12:
I managed to replicate your hang! Turns out that when the web socket is told to run_forever, with no ping timeout, and your network drops, it hangs on listening on the socket, and will never exit. v2.11 should fix the issue. Fingers crossed anyway

tdub415, 2018-07-15:
I haven't had any more drops/offline issues since 2.11, I think your fix worked :)

cameroncros, 2018-07-18:
I've had something similar unfortunately :/ Will investigate and see what can be done

cameroncros, 2018-07-31:
@tdub415 Have you had any more issues with the disconnecting? I did add some more error handling, but i haven't reproduced the hang yet.

tdub415, 2018-07-31:
It's seemed to be pretty stable from what I've noticed since the past couple of updates.  I haven't noticed it being down, but I haven't had to check pause/stop a print lately either, so it may have gone down and me not noticed.