cameroncros/OctoPrint-DiscordRemote #20: geting the ip of the net

deanyarongit, 2018-07-02:
hey i was wondering if there is an option to get the ip of the net from the discord channel.

cameroncros, 2018-07-02:
There isn't currently, but certainly a possible feature. Depends what IP your after, if you want our local network IP, should be easy. If you want the internet IP, might be harder?

cameroncros, 2018-07-02:
I added the IP address to /status. It will try to guess the IP address, but it will only be the local network IP.

deanyarongit, 2018-07-03:
hey tnx was hoping for the outside ip  not local

cameroncros, 2018-07-03:
Hmmmm, no idea how to do that, will have a think about it, but I don't know if it's possible without an outside service

cameroncros, 2018-07-06:
Added ipgetter so that the external IP address as well as the local. Will be in next release.

closed 2018-07-06 by cameroncros