cameroncros/OctoPrint-DiscordRemote #24: /timelapse command to list all timelapse files

cameroncros, 2018-07-07:


OctopusProteins, 2018-08-09:
I don't know if you were still looking into this functionality but this could probably easily be done using this built-in API: http://docs.octoprint.org/en/master/api/timelapse.html

cameroncros, 2018-08-09:
It can be done with no issues, but discord has a file upload limit of 8MB, which makes it somewhat restrictive. If you want to go ahead and do it, i wont stop you, but I personally aren't as interested.

OctopusProteins, 2018-08-09:
I am thinking to have the bot just send the link to the file download, since Discord has that file size limit.

cameroncros, 2018-08-09:
Not a bad workaround, I like that. Maybe upload the file if it is <8mb, and link to it if it is larger?

OctopusProteins, 2018-08-09:
Sure sounds good.

OctopusProteins, 2018-08-09:
Fixed by #35

cameroncros, 2018-08-09:
Ideally, in the future, make a separate branch and PR for separate features. I'll review and try merge sometime today.

OctopusProteins, 2018-08-10:
Fixed by #36

commit 05dd3522 (2018-08-10): Added /timelapse. Fixes #24.
Optimized variables in enclosure plugin.
commit 186988b0 (2018-08-10): Revert "Added /timelapse. Fixes #24."

This reverts commit 05dd3522d0196b9940cae3f7c6e9b2443916fffc.