cameroncros/OctoPrint-DiscordRemote #26: No bot shows up

lildevilx, 2018-07-12:
Hi, i'm having some issues getting the bot to even show up on my channel.

i've made sure that the token id and channel id are correct.
as well I seem to be getting some errors in regards to the discord remote plugin, please see https://pastebin.com/e6GTnH1Q

Thanks!

jon012198, 2018-07-12:
looks like it dose not have permissions in you server

cameroncros, 2018-07-12:
Agree with jon, your bot does not seem to have access. Have a look at the section "Add your new Discord bot user to your Discord server/guild" in https://github.com/Chikachi/DiscordIntegration/wiki/How-to-get-a-token-and-channel-ID-for-Discord

cameroncros, 2018-07-13:
You need to allow your bot to connect to the server. Click the link above, and scroll to the section "Add your new Discord bot user to your Discord server/guild". Follow the instructions there.

lildevilx, 2018-07-13:
o that was silly of me to completely miss that section, Thanks for your help and your updates!

closed 2018-07-13 by lildevilx