cameroncros/OctoPrint-DiscordRemote #29: ip address

jon012198, 2018-07-18:
ip get set back to on when it updated to a new version  when it sends this message  I just woke up! What are we gonna print today?
Local IP: 172.17.62.26 External IP:xxx.xxx.xxx.xxx when i have it set to just local ip

cameroncros, 2018-07-19:
So, the wake up message is seperately controlled. I did change the default, but didn't think that it would update for everyone? Change your startup event to not contain the external IP address.

jon012198, 2018-07-21:
Got it fixed once i removed the ext ip but look like it overrides you settings when it updates

cameroncros, 2018-07-21:
It shouldn't do that, will investigate though

cameroncros, 2018-07-23:
Okay, Did some investigation:
1. If you have NOT changed the message from the default, and the plugin updates the default, you will get the NEW default.
1. If you HAVE changed a message, and then update the plugin, you do NOT get the new default.
So, I think its all working as expected. I will try not to change any of the messages from the defaults in the future, unless there is a very good reason for it.

closed 2018-07-23 by cameroncros