cameroncros/OctoPrint-DiscordRemote #41: Feature request: custom commands

lildevilx, 2018-08-25:
Hi, was wondering it is possible to run customs commands that would send a bash command for raspberry Pi users to shut off the printer or what ever relays they have?

cameroncros, 2018-08-26:
It certainly is possible.

If you have [PsuControl plugin](https://plugins.octoprint.org/plugins/psucontrol/) installed on your octoprint system, you can control it from discord with `/poweron`, `/poweroff`, `/powerstatus`
Or, the [Enclosure plugin](https://plugins.octoprint.org/plugins/enclosure/) has some integration with DiscordRemote, but its limited to turning GPIO pins on and off.

Another possibility would to add support for the [system commands](http://docs.octoprint.org/en/master/api/system.html#list-all-registered-system-commands), which might achieve your goal?

I would be against adding something like:
`/bash sudo poweroff` 
As it would allow anyone on your discord channel to run arbitary bash commands on your pi. But, if it truly is something you want, I can do it and leave it disabled. (Enabled by manually enabling a setting in the octoprint settings)

Which of these would you prefer? I strongly suggest you look into the PsuControl and Enclosure plugins first, as they will work already, otherwise the system commands should be doable. Least preferred is any kind of arbitrary bash execution.

lildevilx, 2018-08-26:
I'll look into enclosure plugin as it is the gpio i do want to control. however i really wouldn't mind system commands :)
i've locked down my discord channel so it wouldn't be too bad + you have it so only certain people can command the bot anyways.

Would it be possible for it to post the result? IE doing ls, will send you the list of items?

cameroncros, 2018-08-26:
Basically none. If you have the enclosure plugin installed, you can use `/help` to see the available commands. 

`/outputon N` where N is the id number from the enclosure plugin, will turn on an output. Similarly, `/outputoff N` and `/outputstatus N` turn off and give the status of the order given.

You can only switch GPIO pins, you can't trigger scripts (limitation of the enclosure plugin)

cameroncros, 2018-08-31:
System commands are now done. See v2.22

commit d6f2e537 (2018-10-09): Fix #41: Allow multiline status messages
closed 2018-08-31 by cameroncros