cameroncros/OctoPrint-DiscordRemote #42: Percent complete off

tdub415, 2018-08-30:
Haven't been printing with octoprint lately because I was having some errors with it, but started it up again earlier this evening.  I noticed that discordremote's reporting % complete wasn't lining up with octoprint's % complete shown (example i got the 70% message while octoprint was showing 87%)

tdub415, 2018-08-30:
and the 90% just now while octoprint was showing 95%

cameroncros, 2018-08-30:
Do you have a particularly slow internet connection? Are these particularly fast prints? Perhaps the messages are just arriving late/slowly?
The plugin does not poll for the percentage, it is triggered by octoprint itself. As soon as Octoprint says its at N%, the plugin sends the percentage message. There is no way for the plugin to send it any faster or earlier. I am not sure what can be done here. 
If the percentage messages were early, that is a problem, but still outside the control of this plugin?

tdub415, 2018-08-30:
My connection speed is fine, I don't think that's the problem.  I'll do some more personal testing and see if I can narrow it down some.  It might be the PrintTimeGenius plugin messing with things, I know it was causing some issues early on with some things for me, maybe it's displaying it's own % complete status and not octoprints?  I'll try printing again with PrintTimeGenius disabled and see if i get the same non-synced status updates.

cameroncros, 2018-08-30:
Hmm, would not surprise me the PrintTimeGenius is messing with how it is reported. Does seem more like an octoprint/PrintTimeGenius bug though? Might need some investigation

tdub415, 2018-08-31:
So I think it must be something with print time genius.  My print is currently showing 82% via octoprint, but when I did a /status, the status is showing 74%

Genuis must be overriding both the time estimate and the % complete from octoprint base with it's own estimates.  Not sure if it would be possible, but do you think there would be a way to capture genius's time/% (or have an option to use it over octoprints)?  Not sure how many people use printtimegenius, but it's soooooo much more accurate than octoprint's estimations, and I love it.

cameroncros, 2018-08-31:
For doing /status, it may be possible, but for the percentage notifications might not be doable. I'll have a look into it over the next few days, but I can't promise anything unfortunately

tdub415, 2018-08-31:
it's not a huge deal if it'd be something complicated to code .I just wasn't sure if it would be a simple thing to just have a point to a different variable or something, I'm more just glad to know what's causing it

cameroncros, 2018-09-02:
I think that it might be best if you report this bug to the PrintTimeGenius dev. They would have a better understanding of whats happening underneath, and perhaps they will be able to fix it, or narrow down where the bug exists (I suspect octoprint, but I cant prove it easily). If you report it to PrintTimeGenius, please post the issue link here so I can keep an eye on it, and help if needed.

closed 2018-09-02 by cameroncros