cameroncros/OctoPrint-DiscordRemote #43: Add Name of printer

rhys100, 2018-09-04:
Is it possible to have the printer name post in discord? I have 2 printers posting to same channel but doesnt say which is which etc

cameroncros, 2018-09-05:
Should be doable, would you be happy with the /status command having a field for the printer name? Or something else?

Embeds have an author field, I could try put the printer name there? 

I have no idea how to get the printer name, but I'll look into it. 

Also, bear in mind that both discord bots will listen to the commands you send, which will cause them to both respond. There is no way around this, so I suggest you create different channels for each printer. Also, you could create 2 discord bots, one per printer, and then you can see which one is which.

rhys100, 2018-09-05:
I was thinking you could pull the title from appearance-->title and just have it at the top of the discord msg. 

so when it posts it could be like this

"PRINTER NAME"
:printer:️ I've started printing
file name

You could just have a checkbox under message settings to enable title or not.

alternatively I can set up seperate bots for each printer, but thats rather tedious for 8 printers

cameroncros, 2018-09-05:
Okay, Getting the title from the server is easy, and i can put that into the author field, which i'll do, but I still think you have an issue with sending commands to your printers, unless you are creating new channels for each printer. I'll get the author change done in the next few days.

cameroncros, 2018-09-11:
See release v2.23. I still don't suggest you do what you are planning, but hopefully that might help.

closed 2018-09-11 by cameroncros