cameroncros/OctoPrint-DiscordRemote #44: Discordbot problems

subsoniq, 2018-09-11:
Hi, the part about creating a new discordbot and giving it the right rights to write something in the channel is poorly covered. I have added my bot to my discord channel and everything looks green and ok. It joins my discord channel but I am unable to get it to write anything in the channel. I have doublechecked all the channel settings and tried giving the bot all kinds of different bot permisions. Even tried admin =8, but it still doesnt want to write anything in the channel. Any tips to checking what could have gone wrong? 
Here is the octopi error log
https://pastebin.com/EUJTBPbT
(for the record i have changed my discord bot token after pasting this log

cameroncros, 2018-09-11:
The documentation is quite weak, and really could do with some work.

Regarding your issue, I think you have not got your channel configured correctly

subsoniq, 2018-09-11:
I am using that id channel number as well but still it will not say anything

cameroncros, 2018-09-11:
You are using that exact channel number? Or the equivalent one for your server? The error discord is returning is that the channel ID doesn't match a real channel.

cameroncros, 2018-09-13:
@subsoniq if you don't reply to this thread soon, I'll assume that you've solved your problemand close this ticket

closed 2018-09-15 by cameroncros