cameroncros/OctoPrint-DiscordRemote #46: Having issues installing.

znelson710, 2018-10-01:
I have tried restarting and rebooting but I cannot get the plugin to install. 
Installation gets to "Running setup.py install for Pillow: started" and then freezes. After a little while I gt this error message: Error!
Could not install plugin, unknown error, please consult octoprint.log for details

I have attached my log.
[plugin_pluginmanager_console (1).log](https://github.com/cameroncros/OctoPrint-DiscordRemote/files/2432330/plugin_pluginmanager_console.1.log)
[octoprint (3).log](https://github.com/cameroncros/OctoPrint-DiscordRemote/files/2432331/octoprint.3.log)

cameroncros, 2018-10-01:
Interesting. Can you download https://files.pythonhosted.org/packages/0f/57/25be1a4c2d487942c3ed360f6eee7f41c5b9196a09ca71c54d1a33c968d9/Pillow-5.0.0.tar.gz on your PC? 
Are you able to SSH into your octopi instance, and type the following:

```
source .octoprint/bin/activate
pip install https://files.pythonhosted.org/packages/0f/57/25be1a4c2d487942c3ed360f6eee7f41c5b9196a09ca71c54d1a33c968d9/Pillow-5.0.0.tar.gz
pip install https://github.com/cameroncros/OctoPrint-DiscordRemote/archive/master.zip
```

znelson710, 2018-10-01:
I can download the folder on my pc.
I can ssh in but typing in:
source .octoprint/bin/activate
gets me:
-bash: .octoprint/bin/activate: No such file or directory

the other commands get me 
-bash: pip: command not found

I have pip installed. I just upgraded it and still get the error.

cameroncros, 2018-10-01:
Sorry, I messed up, the path should be:
```
source oprint/bin/activate
```

cameroncros, 2018-10-02:
Is it downloading the tar.gz successfully?
Do you have `gcc` installed? (`which gcc`)
Are you saying that the pi drops it's network connection just after downloading, and before it starts compiling?
Is your pi's power supply a good quality one? Perhaps when it starts compiling pillow your pi is running out of power and resetting?
Check your `uptime` when you get back in, if your pi rebooted, it will be set back to a small number

znelson710, 2018-10-02:
It was the power supply. I usually have it on a smaller one but I plugged it into a beefy quick charger and it was able to chug along. Thanks! Now to the google control

closed 2018-10-02 by znelson710

CameronWork, 2018-10-02:
No problem, glad I could help.

cameroncros, 2018-10-02:
No problem, Glad I could help.