cameroncros/OctoPrint-DiscordRemote #47: Set up instructions?

znelson710, 2018-10-02:
I have my bot token and channel ID in the plugin. For the base URL I used the local IP as the pi is not port forwarded. The gamepad icon is green and my bot says online in my discord channel but nothing is sent or received from either direction. 
Is there any special permissions when setting the bot up so it can read/write in the channel

disregard the top commands that is the google assistant bot I was testing out.

[image]

cameroncros, 2018-10-02:
Check that your channel ID is a valid channel ID (Should be a number)? If you click the gamepad icon, you can send a command to the bot, which will get send to the output to discord. Send a `/help` command, and see if anything gets printed.

znelson710, 2018-10-02:
I was using the ID from the "command center" channel. I went to the #commands settings and copied something from that and it works now! I've never really used discord before. But with a file that is just bed and nozzle temps I can tell my printer to warm up with my voice!

Thanks for all the help! I should be all set now!

cameroncros, 2018-10-02:
I'm very curious about the IFTTT integration your working on, seems like a very good idea for a standalone plugin. If I ever have some spare time and I'm bored, maybe I'll look into writing a IFTTT plugin.

znelson710, 2018-10-02:
Will do. A basic preheat command would be useful rather than storing a "print" with heat code on the pi. If think in having issues with the remote bot not listening to the Google assistant bot. I know I have your plug in configured because I can send commands and it listens. Can your plug in listen to bots?

znelson710, 2018-10-03:
I will see about another way to make a non bot post in discord as I have never really played around in the coding of things like this... just getting services to talk to each other with the tools already created.

cameroncros, 2018-10-03:
I'll have to investigate that, might take a day or two. You could try the web API, which is how the web interface sends commands. I don't know what the URL is http://serverurl:port/plugin/discordremote

You send a POST message containing JSON, of the format:
```
{
                    command: "executeCommand",
                    args: command_string
 }

```

cameroncros, 2018-10-03:
https://github.com/cameroncros/OctoPrint-DiscordRemote/blob/master/octoprint_discordremote/static/js/discordremote.js

znelson710, 2018-10-03:
Since I don't have the pi open to the public net I don't believe the Google Assistant commands will reach it. The flow of my set up is google command -> IFTTT to discord via webhooks bot -> command read by discord bot plugin -> octoprint executes command. 

It works since the plugin accesses discord on its own and handles the api requests and stuff. At least this is my theory. I want to say I read somewhere about the pi needing to be on the public net as IFTTT requests will come from outside the network if you were to use IFTTT to make a webhooks api request directly to the pi.

cameroncros, 2018-10-03:
Thats probably true, I will look into removing the bot restriction. If you want to try test it on your own, ssh into your pi, and go to oprint/lib/site-packages/OctoPrint-DiscordRemote/octoprint_discordremote/

There should be a file called discord.py, with the content: https://github.com/cameroncros/OctoPrint-DiscordRemote/blob/master/octoprint_discordremote/discord.py#L248

Go to line 248 and remove/comment out:
```
        if "bot" in data['author']:
            # Only care about real people messages
            return
```
Then restart the octoprint. Hopefully that will work.

znelson710, 2018-10-03:
OK. I will give it a shot. I was unsure how/where to go in and comment out that bot check. I can see how the bot check can be useful to ensure someone else hasn't hijacked a printer with a bot.

Will edit this comment with results. Thanks again for all the help with getting this set up!

cameroncros, 2018-10-03:
The bot check was because originally, the bot would respond to any command it didnt understand with the help output. Which meant that the bot would see its own message, and then send a message, and get stuck in a loop. It no longer does that, so it should be fine.

znelson710, 2018-10-03:
Btw are you on reddit?

cameroncros, 2018-10-03:
No, I dont have an account. Why?

znelson710, 2018-10-03:
I was going to post my setup and show off the voice control and credit you and your plugin. I will post the plugin page on the octoprint repository website.

cameroncros, 2018-10-03:
Ooh, would definitely be interested in reading what you post. Please send a link to the post.

znelson710, 2018-10-03:
Will do!

znelson710, 2018-10-03:
Reddit post!
I did the main post as a link to screenshots and then the explanation in the comments.

https://www.reddit.com/r/3Dprinting/comments/9l0lf2/ok_google_lets_get_printing/

cameroncros, 2018-10-03:
Thanks! Looks great. I have created a new release that allows bots to talk to discord. If you get prompted to update it should be safe and shouldn't break your existing setup.

commit 2d668a75 (2018-10-03): Fix #47: Allow bots to send commands via discord
commit 388df8ed (2018-10-03): Fix #47: Allow bots to send commands via discord
closed 2018-10-03 by cameroncros