cameroncros/OctoPrint-DiscordRemote #50: [Feature] Pause for [x] Seconds before Snapshot

TheSquall, 2018-10-20:
Can we have a way to wait for x seconds before taking a snapshot? This I think would mostly be for the "done" photo. My photos are being taken immediately after the 100% is hit, which means my extruder may be blocking part of the print and so I can't see a good snapshot of the part. Just a thought. Thanks!

cameroncros, 2018-10-20:
Why not add a GCODE move at the end of your print to send the extruder out of the way? Alternatively, adjust your camera angle? I don't want to add this feature, as I don't think it's valuable for everyone. Being python, you could always modify the plugin locally for yourself, it's a minor change

TheSquall, 2018-10-20:
My printer does have the gcode as mentioned, but DiscordRemote likes to take the picture before the movement is completed. For instance:
[image]

I was using the Prusa supplied gcode for this, so I haven't edited anything from it. I could probably add in a wait command or something similar for that, but i wasn't sure if it's something others have seen.

cameroncros, 2018-10-21:
Thats very interesting that octoprint is getting triggered before the move has finished. Without knowing more about the GCODE, I cant really help any more. I still dont want to add a delay to the plugin, but if you want to do it yourself, try:

https://github.com/cameroncros/OctoPrint-DiscordRemote/blob/master/octoprint_discordremote/__init__.py#L312
```
if int(data["progress"]) == 100:
    time.sleep(1000)  # 1000 ms == 1 s sleep
```

cameroncros, 2018-10-25:
I'm gonna close this issue. Sorry I can't help, but I think it's just too niche to add into the plugin.

closed 2018-10-25 by cameroncros