cameroncros/OctoPrint-DiscordRemote #51: Notifications with image always called “snapshot.png”

Patronics, 2018-10-24:
In Octorant (that this plugin is based on/inspired by), whenever you get a notification (including those with snapshots) the notification actually displays the message of the notification, such as the percentage of progress the printer has made. However with this plugin whenever the message includes an image the notification (on the Discord iOS app anyway) just says “snapshot.png”, not the actual text of the status message.

cameroncros, 2018-10-24:
Yeah, I am aware of that, but I am not really sure what the fix would be. This is a problem specific to discord embeds I think.
I can do some investigation, but I think this may be a discord problem, not a plugin problem?

cameroncros, 2018-10-25:
I decided to just split the snapshot out from the second main message. Best option for now.

commit aacbc474 (2018-10-25): Fix #51 Send snapshot seperately to notification, as discord mobile notifications only say snapshot.png
closed 2018-10-25 by cameroncros