cameroncros/OctoPrint-DiscordRemote #52: Long tooltip breaks TouchUI navigation

blazze11, 2018-11-03:
The long tooltip of the navbar icon breaks the navigation on touchscreen devices with the touchui plugin.
Maybe change it to show only DiscordRemote.
[image]

cameroncros, 2018-11-03:
Hmm, interesting. This kinda feels like a bug in TouchUI, rather than this plugin. Have you asked them whether that dev is willing to look into it? If they don't care/don't want to fix it, I am more than happy to shorten the message

blazze11, 2018-11-03:
I will report it to the touchui dev. Thanks for your response.

cameroncros, 2018-11-04:
Tracked by https://github.com/BillyBlaze/OctoPrint-TouchUI/issues/253

cameroncros, 2018-12-12:
@blazze11 is this something that still bothers you?

blazze11, 2018-12-12:
Yes

cameroncros, 2018-12-13:
Righto, I'll shorten the message for now. Seems the other project is not too interested in fixing it.

commit e1a422d6 (2018-12-16): Fix #52 Made the title for the navbar shorter to prevent TouchUI overflow.
closed 2018-12-16 by cameroncros