cameroncros/OctoPrint-DiscordRemote #54: /status commands not working

Joshua-Roe, 2018-12-14:
running octoprint on puppy linux 14.04 thar. Everything appears to be set up correctly both on octoprint and discord text channel, all commands work apart from /status, tried from discord and from octoprint page.

cameroncros, 2018-12-16:
Hi Joshua, Can you please get the logs from your octoprint instance and attach them to this ticket? I dont have enough information to debug your problem.

Joshua-Roe, 2018-12-18:
@cameroncros which logs do you mean as I cannon find a log file for discordremote

cameroncros, 2018-12-19:
@Joshua-Roe Discord remote logs get lumped into the main octoprint logs, can you get me those? Give them a quick read over yourself to ensure there is no private information though.

Joshua-Roe, 2018-12-19:
[octoprint.log](https://github.com/cameroncros/OctoPrint-DiscordRemote/files/2695267/octoprint.log)

Hi here is the main octoprint log file,
I tested the command before saving and notice the error at end of file.

cameroncros, 2018-12-20:
Very interesting. Give me a few days to analyse your issue and I will try get it fixed. In the short term, you could try going to the settings for DiscordRemote, and disabling the external ip address lookup. Something there is buggy for you.

cameroncros, 2018-12-23:
@Joshua-Roe I am not certain, but it seems that your python ssl module is broken. You may need to reinstall it. Python ssl should be built into the system, so you might need to use your linux package manager. 

Side note: Puppy Linux is up to 7, and 14.04 is an Ubuntu version number. Which OS are you using? Also, Ubuntu 14.04 was release 4 years ago, you may want to get to update your version?

cameroncros, 2019-01-04:
@Joshua-Roe Im gonna assume this is closed. If you still have issues, please re-open this ticket and supply some logs. I can also remotely debug if you are comfortable with that.

closed 2019-01-04 by cameroncros