cameroncros/OctoPrint-DiscordRemote #55: Bot will not respond to commands

strb857, 2019-01-17:
I am probably missing something obvious, but I can't get the bot to respond to commands in discord. Everything else is working. I can see the bot on the channel, I can send commands from octoprint which shows up in Discord, but not send commands from discord itself.

Any ideas?

I see there have been issues raised on this in the past, but I couldn't find a solution there.

cameroncros, 2019-01-23:
@strb857 @ethanfel Sorry for the delay, i was away on a holiday.

[image]
Please try clicking the controller icon, and typing /help into it. Let me know if the discord bot responds to the command in discord. 

Also, make sure you have the correct discord channel value.

ethanfel, 2019-01-23:
Hi, I hope you add nice holiday.

This way it work, the bot send the help message.

It doesn't react to command send from discord.

cameroncros, 2019-01-23:
Interesting, can you send me your logs?

ethanfel, 2019-01-24:
after a few days, everything seems to work. XD It may habe been a discord issue.

cameroncros, 2019-01-24:
@strb857 do you still need help?

strb857, 2019-01-25:
Sorry for leaving you hanging, been off-line for a while.
Yes, I can still not get it to work. Pretty sure I set it up as expected.

The symptoms are the same as @ethanfel 

- sending command from opctoprint  displays correctly in discord
- typing the same command in discord get no reaction

What logfile would help shed some light on this?

ethanfel, 2019-01-25:
Mine worked for a time but didn't respond after a reboot

cameroncros, 2019-01-25:
Can both of you please send some your octoprint logs?

ethanfel, 2019-01-26:
sorry for the delay i was away from home.

https://pastebin.com/rqk22Kqu

ethanfel, 2019-01-27:
> Had the same problem. How I made it work:
>
> With dev mode enabled, right click on your discord profile on the
> right-side list with server members and click copy ID. Then paste it in the
> whitelist field of discord remote. For some reason it doesn't allow for an
> empty whitelist.

it seem to have worked for me, it respond to command after a reboot now

cameroncros, 2019-01-27:
@frezap I don't know what is going on for you, but the bot certainly does work with an empty allow users feild. It just allows all users.

ethanfel, 2019-01-27:
https://pastebin.com/Dz8RMZMY

cameroncros, 2019-01-27:
@ethanfel it is definitely seeing some messages, but it doesn't look like it saw any messages from you. Did you send a /status command via the web interface to get it to print that status message? If you sent it via discord, you should see a "Message received: ..." Line at contains whatever text you sent to the bot.

cameroncros, 2019-02-02:
Is this still an issue people are having? Without more logs, I cannot help any further. And given it is working for me, and my test server, unless someone has more useful logs, i'll have to close this ticket.

ethanfel, 2019-02-02:
seems to work fine for me now. Sometimes it doesn't respond but most of the
time it does.

frezap, 2019-02-02:
Works since I installed Octoprint on a beefier machine. Thanks for the great Plugin, love it a lot!

vfrdirk, 2021-02-24:
2021-02-24 16:05:30,590 - octoprint.plugins.discordremote - INFO - Received HEARTBEAT_ACK message
2021-02-24 16:05:45,329 - werkzeug - INFO - 127.0.0.1 - - [24/Feb/2021 16:05:45] "[37mGET /?action=stream&1614182745129 HTTP/1.1[0m" 200 -
2021-02-24 16:06:11,576 - octoprint.plugins.discordremote - INFO - Heartbeat: {"op": 1, "d": 48}
2021-02-24 16:06:11,651 - octoprint.plugins.discordremote - INFO - Received HEARTBEAT_ACK message
2021-02-24 16:06:41,120 - werkzeug - INFO - 127.0.0.1 - - [24/Feb/2021 16:06:41] "[37mGET /?action=stream&1614182800901 HTTP/1.1[0m" 200 -
2021-02-24 16:06:46,032 - werkzeug - INFO - 127.0.0.1 - - [24/Feb/2021 16:06:46] "[37mGET /?action=snapshot HTTP/1.1[0m" 200 -
2021-02-24 16:06:46,067 - octoprint.server.api - ERROR - Error while executing SimpleApiPlugin discordremote
Traceback (most recent call last):
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint/server/api/__init__.py", line 146, in pluginCommand
    response = api_plugin.on_api_command(command, data)
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint/util/__init__.py", line 1890, in wrapper
    return f(*args, **kwargs)
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint_discordremote/__init__.py", line 346, in on_api_command
    return self.execute_command(data)
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint_discordremote/__init__.py", line 356, in execute_command
    snapshots, embeds = self.command.parse_command(args)
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint_discordremote/command.py", line 78, in parse_command
    return command['cmd']()
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint_discordremote/command.py", line 364, in status
    snapshots = self.plugin.get_snapshot()
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint_discordremote/__init__.py", line 530, in get_snapshot
    return self.get_snapshot_camera()
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint_discordremote/__init__.py", line 563, in get_snapshot_camera
    img = Image.open(snapshot)
  File "/home/pi/oprint/lib/python3.7/site-packages/PIL/Image.py", line 2915, in open
    preinit()
  File "/home/pi/oprint/lib/python3.7/site-packages/PIL/Image.py", line 361, in preinit
    from . import GifImagePlugin
  File "<frozen importlib._bootstrap>", line 983, in _find_and_load
  File "<frozen importlib._bootstrap>", line 967, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 677, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 724, in exec_module
  File "<frozen importlib._bootstrap_external>", line 857, in get_code
  File "<frozen importlib._bootstrap_external>", line 525, in _compile_bytecode
ValueError: bad marshal data (invalid reference)
2021-02-24 16:06:46,109 - tornado.access - ERROR - 500 POST /api/plugin/discordremote (::ffff:192.168.0.166) 1495.43ms
2021-02-24 16:06:52,624 - octoprint.plugins.discordremote - INFO - Heartbeat: {"op": 1, "d": 48}
2021-02-24 16:06:52,687 - octoprint.plugins.discordremote - INFO - Received HEARTBEAT_ACK message

cameroncros, 2021-03-15:
@vfrdirk Are you able to get me a list of the packages installed on your octoprint python environment?
SSH into your octoprint machine:
```sh
source oprint/bin/activate
pip list
```