cameroncros/OctoPrint-DiscordRemote #56: Bot disconnects

lac106, 2019-01-25:
Hi, I just installed the plugins and everything was working fine until I started a print. I was waiting on my 10% notification and when came back a while later, the bot had disconnected from discord, I wasn’t able to send commands and the print was at 18%.

Any ideas? Thanks!

cameroncros, 2019-01-25:
Can you find the logs and attach them to the ticket?

cameroncros, 2019-01-25:
https://discourse.octoprint.org/t/where-can-i-find-octoprints-and-octopis-log-files/299

The DiscordRemote logs will be within the octoprint.log file. Ensure you get the logs for the dates that you had the issue on.

cameroncros, 2019-01-28:
@lac106 if I don't hear from you in the next few days, I'll assume you have solved your problem and close the ticket.

closed 2019-01-30 by cameroncros