cameroncros/OctoPrint-DiscordRemote #57: Progress my time instead of %

Fabrizio3, 2019-01-25:
can we have the bot send the message at a determined time interval? like, i dont want 10 snapshots of a 20min job, but i dont want 2 of a 20h job
so using it by time (for example, every 30 mins) so it sends a message after the specified interval has passed

it would be amazing to get a notification every 30mins (or whatever you set it to) on discord, i monitor my prints while away with it and its very useful, Amazing plugin btw!!

cameroncros, 2019-01-26:
This should be doable I think. Give me a week or so to get to it though, im a little busy with other things atm

cameroncros, 2019-02-02:
@Fabrizio3 The v2.31 release has periodic reporting. Try it out and let me know if it is good enough.

closed 2019-02-02 by cameroncros

Fabrizio3, 2019-02-02:
I really can't thank you enough!!!, I'll try it as soon as I can and let you know!

Fabrizio3, 2019-02-04:
well, it works wonders! thanks!!
it's very convenient now, have set it to 1/2h intervals and  it's not annoying, and also very much adaptive (for small prints  it won't bother me, and  large ones keep me notified!)
probably seems like a simple thing, but the user experience is VERY nice!