cameroncros/OctoPrint-DiscordRemote #58: Status Message separate from snapshot

tdub415, 2019-02-05:
I know a while back in issue #51 you changed it so that it would send multiple messages, one with just the status update, and one with the picture.  Would it be possible to have an option that you could toggle between having the old way and the current way?  I personally preferred getting one message over two, and the current fix always has the picture second, so my notification still shows "snapshot.png" instead of the status.

cameroncros, 2019-02-06:
The entire point of #51 was to make the notification more useful, so if its still not working, perhaps I need to relook at that. Out of curiosity, what platform are you getting the snapshot.png notification and nothing else?

tdub415, 2019-02-06:
I get both notifications just fine, but since the screenshot image comes in second, (and is the most recent) it is the one that's displayed when i look on my watch specifically.  Also getting 2 notification sounds every time on discord through my headset is a bit annoying, lol.

cameroncros, 2019-02-06:
The screenshot should be the first from memory, but I would have to check that. I'll do some research to see if it is possible to disable the notification for a single message (so the snapshot.png is silent), and failing that, I'll add the toggle you wanted.

Just for my benefit, apple watch? I have an android watch, but I haven't really noticed anything weird about the notifications yet (But I dont print nearly as much as I would like)

tdub415, 2019-02-06:
Samsung Gear S3 Frontier in my case, but the screenshot is definitely second in discord for me in the application:
[image]

cameroncros, 2019-02-06:
Hmm, interesting, i'll investigate and get back to you then. Might be a while though.

cameroncros, 2019-02-10:
According to the other developers in the discord api channel, there is no way to not send a notification with a message. I have reported it as a bug that the snapshot.png shows up instead of the title. For now, i think i will continue to think about a workaround.

cameroncros, 2019-03-02:
I have put in another bug report, which should be easily checkable. Hopefully it gets looked at.

cameroncros, 2019-03-02:
https://trello.com/c/isrF8Omu <- discord bug report. Once that is fixed, I'll merge the messages together, and this problem will be solved.

goscicki, 2019-04-11:
I 2nd this. On my Gear S2 it just says snapshot.png and I have to open the app to see what came before the snapshot.

goscicki, 2019-04-12:
I just noticed that sometimes the snapshot comes first and sometimes it comes last, even on android. Perhaps a short delay between sending the snapshot and the message would give enough time to ensure the message is displayed last. After all, it must take longer for the snapshot to be transmitted and processed at every step along the way.

cameroncros, 2019-04-12:
I'll merge the messages together, and hopefully discord will fix the bug with the notification

tdub415, 2019-06-10:
I was brainstorming while at work (slow day) and had a thought.  You could set up the snapshot filename to be "Progress_X%.png" instead of Snapshot.png" maybe, while this wouldn't 'fix' the discord embed notification, it would be a workaround that should allow people who see snapshot.png instead of the message to get the info they need without opening up discord proper.

cameroncros, 2019-06-11:
@tdub415 Great minds think alike! I had the same thought, unfortunately, when i tried to implement it, it completely broke the embeds (I don't recall the exact issue, but it was very broken). Maybe worth trying it again and seeing if I can make some progress though.

commit a1d52e14 (2019-04-29): Merge snapshot and message, pending discord fix #58
closed 2021-10-17 by cameroncros