cameroncros/OctoPrint-DiscordRemote #59: Print Process: done

tdub415, 2019-02-09:
It seems since the latest update, the Print Process: done message is not being triggered.

cameroncros, 2019-02-10:
:( I'll look into it.

cameroncros, 2019-02-10:
My last print, done yesterday had a "done" message printed. I also tested with the virtual server and it also received the "done" message. Do you have any logs for your print that failed to send the message?

tdub415, 2019-02-10:
[plugin_discordremote.log](https://github.com/cameroncros/OctoPrint-DiscordRemote/files/2849330/plugin_discordremote.log)

[plugin_discordremote.log.2019-02-06.log](https://github.com/cameroncros/OctoPrint-DiscordRemote/files/2849331/plugin_discordremote.log.2019-02-06.log)

I notice after looking at these two most recent logs, it seems to be listening to other channels on my discord as well, hence all the warframe alerts being in the log.

cameroncros, 2019-02-11:
That's normal, the bot (as well as a normal client) is notified of all messages on all channels it's allowed (by discord) to see. The plugin limits the channels that it actually responds to by itself.

cameroncros, 2019-02-11:
Nothing about your logs suggests anything is wrong, except that the message is not getting sent. I have added a little more logging, update your plugins and see if it fails on your next print.

Vashidu, 2019-02-11:
I actually have this issue too, I'm printing a thing now, and will grab logs when it is done.

Vashidu, 2019-02-11:
hmm, it seems to have sent the message on this print, let me try a few more prints to see if it still does.

Vashidu, 2019-02-11:
Next print didnt send the finish. here is the attached log. 
[plugin_discordremote.log](https://github.com/cameroncros/OctoPrint-DiscordRemote/files/2853008/plugin_discordremote.log)

cameroncros, 2019-02-11:
Thanks, thats very helpful. It seems that the plugin is not receiving the "done" event from octoprint. I wonder if this is an octoprint flaw? Will investigate further anyway.

tdub415, 2019-02-11:
@Vashidu do you by chance use the ProgressBasedOntime plugin, to swap octoprint to display the % based on print time instead of the % of the filesize processed?
I've been using it for a while with no issues, until the most recent DiscordRemote update.  I did a print with that plugin disabled and it reported done just fine, but doesn't seem to report it being done with it enabled.

Vashidu, 2019-02-12:
@tdub415 I do progress based on % of filesize i think, i dont have a ProgressBasedOnTime plugin.

tdub415, 2019-02-12:
Ok, just wanted to rule that out, I did just have a print finish that sent the done message (with the aforementioned plugin reenabled)

tdub415, 2019-02-12:
I believe your latest release for the added logging didn't increment the version number properly.  The terminal view showed it downloading and installing the 2.32 package, but when it completed it said version 2.31 installed (and it's showing version 2.31 in the plugin manager) and thus keeps prompting to update again.

cameroncros, 2019-02-12:
Yup, fixed it, sorry about that. not sure what happened there

cameroncros, 2019-02-12:
I have updated my instance, and ran a print overnight, which successfully sent the "done" message. I have no idea what is wrong for you guys, but I suspect its unrelated to DiscordRemote?

cameroncros, 2019-02-13:
Yup, I am seeing that as well, not sure what happened. I will fix

cameroncros, 2019-02-22:
@Vashidu @tdub415 do you still have this issue?

tdub415, 2019-02-22:
Yes, after updating to the latest version, I'm still not getting print done messages consistently

cameroncros, 2019-02-22:
Can you give me a list of plugins installed? I still can't reproduce it on my end, which makes it impossible to fix

tdub415, 2019-02-23:
Action Command Prompt Support
Announcement Plugin
Anonymous Usage Tracking
Application Keys Plugin
Backup & Restore
Core Wizard
CuraEngine (<= 15.04)
*Discord Remote (2.33.0)
Discovery
*DisplayProgress(0.1.2) 
*FileManager (0.1.3)
Force Login
Logging
Pi Support Plugin
Plugin Manager
Printer Safety Check
*PrintTimeGenius Plugin (1.3)
*ProgressBasedOnTime Plugin (0.1.2)
*Prusa Mesh Leveling (0.3.0) (this one was added after the problem started occuring, so fairly sure it isn't the cause)
Software Update
Virtual Printer

Most of these are ones that come standard with octoprint, I have put asterisks in front of the ones that I maunally added.

Every now and then I get a 100% message on a print, but usually I don't.  The only plugins I've added in the past month were Display Progress and the Prusa mesh leveling.  I think I installed Display Progress around the time I updated discord remote and started having this issue.

cameroncros, 2019-02-23:
Can you try disabling ProgressBasedOnTime? At least testing on a virtual server, having that plugin enable completely breaks the VIRTUAL printer. It also does mess with how the progress is calculated, which means that it affects how the events that DiscordRemote uses are generated.

tdub415, 2019-02-23:
I can give it a try again without it, but that was the only way I could get the displayed progress (and reported progress from Discord Remote) to sync up with PrintTimeGenuis's estimations

cameroncros, 2019-02-23:
If it does affect the "done" event, you'll just have to report this bug to them, there is nothing I can do to fix it on my end. If DiscordRemote doesn't receive a "done" event, it can't send a message

tdub415, 2019-02-24:
My last print (with progressbasedontime disabled) did not send the done message either.  This leads me to believe it's not that plugin causing the issue.

cameroncros, 2019-02-25:
Thats good i guess, but still doesnt explain what is causing the issue. I'll try load my printer with all of your plugins and see if I can replicate it.

tdub415, 2019-02-25:
If it helps, I also tried a print with Display Progress (what I thought might be causing it) disabled, but still didn't get the "done" message :(

cameroncros, 2019-02-28:
Hmm, interesting. Are you printing from the SD card? Also, what type of printer/firmware do you have?

From Octoprint Changelog:
      * [#2335](https://github.com/foosel/OctoPrint/issues/2335) - First throw at detecting if a print from the printer's SD was started outside of OctoPrint via the printer's control panel. Requires some specific requirements to be fulfilled by the printer's firmware to function properly:
      * Firmware must send a "File opened: ..." message on start of the print
      * Firmware must respond to an immediately sent `M27` with `SD printing byte <current>/<total>`
      * Firmware must stay responsive during ongoing print to allow for regular M27 polls (or push those automatically) or M25 to pause/cancel the print through OctoPrint.
      * Firmware must send `Done printing file` or respond to `M27` with `Not SD printing` when SD printing finishes (either due to being done or to having been cancelled by the user).

tdub415, 2019-02-28:
I have a prusa i3 Mark 3 and all prints were started From within octoprint
I am not printing from the local SD to the printer

cameroncros, 2019-02-28:
Well, i guess that rules out a broken firmware, i think reporting this up to Octoprint itself is the best bet. I can keep debugging, but i am kinda doing it blindly at this point.

tdub415, 2019-02-28:
here's a snip from my last prints octoprint log at the time the print finished https://pastebin.com/pkf7GLfZ

I'm not entirely sure what's going on in the log, but it almost looks like the bot is disconnecting and reconnects after the print finishes, polls for status and sees it as operational, missing the 'done' event, though i'm not sure what might be causing it to close the websocket.

Is there any more thorough logging I can enable to help try to track down the issue?

cameroncros, 2019-02-28:
Actually, thats very interesting, you might be right about the issue being on my end. I'll have a fix for that bug later tonight

cameroncros, 2019-02-28:
@tdub415 @Vashidu Update your discordremote plugin to v2.34 and see if that helps.  Sorry for the delay fixing it.

closed 2019-02-28 by cameroncros

tdub415, 2019-03-01:
I'm 2 for 2 on "done" messages after updating to 2.34, I will continue to
monitor and let you know if there are any issues.

commit 8f353503 (2019-02-28): Fix #59 closing invalid periodic thread breaks done message