cameroncros/OctoPrint-DiscordRemote #61: [Feature Request] Add a person as a watcher

EragonShade98, 2019-02-11:
Would it be possible to add someone that can only use snapshot and status but cannot upload and start prints?

Thanks!

cameroncros, 2019-02-12:
@EragonShade98 That is certainly possible, however, I am not sure its something that I want to get into. Permissions can get very complicated very rapidly. I'll leave this open, and I might do it in the future, but it wont be a priority. If you are capable and want to attempt this yourself, I will definitely be happy to merge it though.

cameroncros, 2019-03-03:
@EragonShade98 I have made an attempt at this, are you interested in testing it before I release it?

EragonShade98, 2019-03-03:
Definitely.

cameroncros, 2019-03-03:
I'll clean it up, and work out how to give it to you to test. Will be a few days.

EragonShade98, 2019-03-03:
Awesome. Keep me posted.

cameroncros, 2019-03-04:
If you want to try it, either:
```pip install git+https://github.com/cameroncros/OctoPrint-DiscordRemote.git@DR-61```
or copy that url into the web interface and install from there.

EragonShade98, 2019-03-06:
Got it, thanks. I'll install it tonight.

cameroncros, 2019-03-09:
@EragonShade98 how's it working out?

megasaturnv, 2019-03-14:
I, too, would love to see this feature. Been testing it out on a virtual machine today and it works really well.

If I had to offer any feedback, I'd suggest you explicitly state if the prefix ('/' by default) is needed in the instructions when adding the list of allowed commands. As you're probably aware, you don't need the '/' in the list of allowed commands (both with and without works). It's probably best to say not to include it, so this covers users who have changed the command prefix.

This was tested on a brand new install of Octoprint with no other plugins. Next test is to uninstall this version of DiscordRemote, clear any configs, install the 'master' DiscordRemote and then upgrade to this version.

megasaturnv, 2019-03-15:
Problem: File upload via Discord doesn't work in this version.

Regarding the upgrade process from master, discordremote: allowedusers: '\<id\>' is retained in the settings, but ignored in this version (as I can see from the commits). This means an upgrading user who has configured only themselves to have access to DiscordRemote would find other users have access in this new version. Ideally users should configure permissions to a channel in Discord, but you can't guarantee it. Would there be any possibility of replacing:
```
allowedusers: '<id>'
```
with
```
permissions:
      '1':
        users: '<id>'
```
if allowedusers: '\<id\>' is found in config.yaml?

If you wanted to add 'permissions': alongside 'allowedusers:', instead of replacing it, you'd need to check if 'allowedusers:' already exists so you don't add multiple 'permissions:' each time. Not sure how well this solution will work with multiple IDs in 'allowedusers:', I'm guessing they're comma-delimited so will require extra work.

megasaturnv, 2019-03-15:
Nevermind, I was wrong, that's why you never read commits on a mobile phone at night... I can see you've already put the effort in to replace permissions - Awesome!

Regardless, permissions aren't added to the config for me - even after a restart. I'll look into why tomorrow. Will probably wipe ~/.octoprint and start fresh.

cameroncros, 2019-03-15:
Yup, the permissions transfer process worries me as well. It needs some time and effort put into it to make sure it all works.

megasaturnv, 2019-03-17:
Did some more testing . Had to comment out line 171 and 172 in __init__.py to disable pdb.

The 'allowedusers' -> 'permissions' setting changeover works on a running instance but the setting is not saved to config.yaml so is lost on a restart. On a restart however, 'allowedusers' is still detected in config.yaml so the plugin intervenes, sends the warning message to discord and runs the changeover code again, essentially putting the settings of the plugin in a state of 'limbo'.

This can be fixed by going to Octoprint's settings > DiscordRemote and clicking Save. Is there any chance this could be done in code? If not (bummer), I guess the best thing would be to mention these instructions in the warning message sent to the discord channel.

Although inconvenient, something to make the plugin safer would be to disable all commands by default instead of allow all by default (__init__.py line 136).

cameroncros, 2019-03-18:
https://github.com/cameroncros/OctoPrint-DiscordRemote/blob/DR-61/octoprint_discordremote/__init__.py#L171 <- Well that is certainly a mistake! I will fix that up. Thanks for the help, I will look into saving the changes, but failing that, disable all by default might be the safest option i think?

megasaturnv, 2019-03-18:
Sounds good. I've been playing around with saving the changes too, but I'm not very experienced with OctoPrint's plugin system. Don't forget - I think file upload via Discord also doesn't work, but that's not as important right now.

cameroncros, 2019-03-18:
yup, I am looking into the fileupload as well.

cameroncros, 2019-03-18:
- [ ] Fix FileUpload
- [ ] Remove 'import pdb'
- [ ] Fix transitioning of permissions, or fall back to disable all commands.

cameroncros, 2019-03-18:
So, testing on the virtual server, it seems to properly transition the settings properly, and save them?

megasaturnv, 2019-03-19:
Just re-tested from an earlier snapshot (Octoprint setup but no plugins installed). Can confirm file upload works for me too - great! However, adding the new 'permissions' section to config.yaml (and removing 'allowedusers' ) is still broken for me. After each restart I keep getting:
> :warning:️:warning:️:warning:️ Allowed users has been changed...
and the config file doesn't change at all.

I've got a spare Pi 1B sitting around. Not powerful enough for printing but should serve for testing this with the Octopi image. I'll try that about a day's time.

Just a heads up, commit 26b8de3 (DR-61 Fixed broken file upload) adds a test.gcode file to the root of this repo. Guessing this was accidental.

cameroncros, 2019-03-19:
Adding the test.gcode was intentional, helps with testing uploading a file. Possibly could exist in a better place, but whatever its there now.

I will try testing on a real server and see what happens regarding the porting of the permissions.
- [ ] Test permissions transfer on a real octoprint instance.

cameroncros, 2019-03-19:
I just tested on my real printer, and it correctly transitioned the the settings across. I went from the current release of DiscordRemote to the latest code in this branch, and it succeeded completely. I wonder if there is something else going on here?

megasaturnv, 2019-03-22:
Sorry for the delayed response - been setting up a 3D printer. However, I think I've figured out the problem

Can confirm porting permissions initially worked on my old Pi B running the OctoPi image. However after changing the DiscordRemote settings to match my virtual machine, the config's stuck in it's old 'limbo' state.

The only change made there was unticking "Show External IP".

After removing `show_external_ip: false` from config,yaml, the porting permissions works fine again. (I could have ticked "Show External IP" via the OctoPrint webpage, but this would have force-saved the new permissions settings to config.yaml on clicking the save button, as mentioned in a previous comment). Might be `show_external_ip: false`'s fault or perhaps other config settings for DiscordRemote in config,yaml. Can't test any more now as the Pi B is very slow, will focus on updating my virtual instance to remove `show_external_ip: false` and then re-checking the permissions change.

If you could double-check the upgrade with "Show External IP" unticked on your end that would be great. A separate (minor) issue I found along the way is unticking "Show Local IP" doesn't seem to do anything - shows the local IP regardless.

megasaturnv, 2019-03-22:
Can confirm permissions changeover works on my virtual OctoPrint installation if `show_external_ip: false` is removed from config.yaml

cameroncros, 2019-03-22:
this is getting confusing, i put `show_external_ip: false` in my virtual config, and it transitioned just fine...

megasaturnv, 2019-03-22:
Test process:
1. Install Octoprint on Debian Stretch following these instructions (with minor changes to change 'pi' username to 'octoprint'). Setup OctoPrint with the most basic of installation settings (I can go into more detail if you want, but basically using reccomended settings with authentication enabled)
2. Install current release of DiscordRemote via OctoPrint plugin manager
3. Setup DiscordRemote: add test discord channel & bot. Only allow my discord user to run commands. Untick show external IP. Test only I have permissions in to run DiscordRemote commands in Discord channel.
4. Snapshot virtual machine to restore back to at a later time
5. Upgrade DiscordRemote via OctoPrint webpage (Settings > Plugin manager > ... from URL > git+https://github.com/cameroncros/OctoPrint-DiscordRemote.git@DR-61)
6. Restart octoprint service

At this point warnings of changover appear in the discord channel after every OctoPrint service restart (I do 3 restarts to be sure), and it doesn't save changes to config.yaml. If I remove `show_external_ip: false` from config.yaml and do a a restart, the warning message appears again but this time changes are saved to config.yaml. After subsequent restarts, the warning message about the changeover doesn't appear again.

I may do some more tests at some point to see if other DiscordRemote settings are affecting this (e.g. disable show local IP, custom discord status messages etc).

megasaturnv, 2019-03-22:
For the OctoPi test, process is the same minus the octoprint installation and snapshotting

cameroncros, 2019-03-22:
Hmm, very bizarre, nothing about that differs to significantly from what i tried. I will try with debian stretch and see what happens. On an entirely unrelated note, I would like to credit you in the README.md, are you okay with that?

megasaturnv, 2019-03-22:
Yeah, it's strange how to the file is written correctly if `show_external_ip: false` is removed. very strange indeed but we'll figure it out.

Sure, I don't expect to be credited but I won't say no to it either so I gladly accept. Thanks!

cameroncros, 2019-03-23:
Testing on Arch linux, but the same way i tested before, and no issues. You wouldnt happen to have some kind of caching going on such that when you do the pip install of the plugin from the URL your getting the old version somehow?

megasaturnv, 2019-03-23:
Nah, don't think so, otherwise the warning message about the permissions changeover wouldn't appear in Discord. I'll get the Pi B out again, install the Octopi image and then write down every step I go through here.

megasaturnv, 2019-03-24:
Running through Octopi setup, DiscordRemote & DiscordRemote upgrade on a Pi Zero this time. It's slow so I'll update this comment as the steps complete:
1. Write octopi 0.16.0 to an 8 GB Micro SD card: `dd if=2018-11-13-octopi-stretch-lite-0.16.0.img of=/dev/sdd`
2.

EragonShade98, 2019-03-26:
Sorry for the late reply. Just got around to trying to install it, and for
some reason the link you have doesn't work. Can you provide me a new
install link? My pi says the following:

HTTP error 404 while getting
https://github.com/cameroncros/OctoPrint-DiscordRemote.git@DR-61Could
not install requirement
https://github.com/cameroncros/OctoPrint-DiscordRemote.git@DR-61
because of error 404 Client Error: Not Found for url:
https://github.com/cameroncros/OctoPrint-DiscordRemote.git@DR-61Could
not install requirement
https://github.com/cameroncros/OctoPrint-DiscordRemote.git@DR-61
because of HTTP error 404 Client Error: Not Found for url:
https://github.com/cameroncros/OctoPrint-DiscordRemote.git@DR-61 for
URL https://github.com/cameroncros/OctoPrint-DiscordRemote.git@DR-61You
are using pip version 9.0.3, however version 19.0.3 is available.You
should consider upgrading via the 'pip install --upgrade pip'
command.Error!Could not parse output from pip, see
plugin_pluginmanager_console.log for generated output

megasaturnv, 2019-03-26:
@EragonShade98 Just to confirm, what command are you running? Should be:
`pip install git+https://github.com/cameroncros/OctoPrint-DiscordRemote.git@DR-61`. I'm guessing you missed off the git+ before https:// :)

EragonShade98, 2019-03-26:
I actually just pasted the url into the web panel, as there's an option at
add plugins via URL. As far as pasting that command into the terminal, it
results in another error as well.

megasaturnv, 2019-03-26:
The error on the terminal is probably because you're not in python's virtual environment for Octoprint. `source ~/OctoPrint/venv/bin/activate' will probably fix that, but let's ignore it for now.

As for the install via Octoprint's web panel, you can follow my instructions from an earlier comment:
> 7. Upgrade DiscordRemote to DR-61 branch: Settings > Plugin Manager > Get More... > ... From URL > git+https://github.com/cameroncros/OctoPrint-DiscordRemote.git@DR-61 > Install

EragonShade98, 2019-03-26:
>    1. Upgrade DiscordRemote to DR-61 branch: Settings > Plugin Manager >

Ah, I see now. You have to include the git+ even in the url box. Thanks for
your help!

cameroncros, 2019-04-01:
Okay, finally confirmed that there is an issue, I will work out what is going on.

EragonShade98, 2019-04-01:
Speaking of issues, there's a bug where if the person says anything, the
bot responds with permission denied, even if it isn't a command being
issued. Is that what you're talking about?

cameroncros, 2019-04-01:
Nope, that is another issue :(
- [ ] Bot responds to all messages with permission denied
- [ ] Settings don't transfer if following megasaturntv's steps

cameroncros, 2019-04-24:
So, my pi with the "not correctly transitioning" settings has somehow transfered them since the time i last tested. I am a bit lost. I think I will make it do a best effort port of the settings, but default to all commands disabled, and then release and merge it. I'll get rid of the notification about the settings, as disabling all commands by default seems reasonably safe.

megasaturnv, 2019-04-24:
I think if you go into the octoprint settings and click save (even without making any changes, if I remember correctly), it will update config.yaml and the new permissions will get written to the config.

I'll get my testing Pi Zero out later and set it to restart every 10 minutes to see if it transitions eventually.

Personally I don't mind the warning it gives in discord since it only shows up after Octoprint is restarted (which is rare for my setup). Perhaps you could change it so it only sends a warning if currently nobody has permissions to do anything? I guess that way the warning won't show if the user's settings transisioned properly first time.

cameroncros, 2019-04-25:
im far more comfortable with the server being completely locked down and no warning than the other way around. I'll change it to disable all commands by default, and call it a day. I'd rather just get this feature out instead of chasing this weird bug

cameroncros, 2019-04-28:
Merged and released.

closed 2019-04-28 by cameroncros