cameroncros/OctoPrint-DiscordRemote #62: Long prints generating multiple progress notifications when a timeout is set

tdub415, 2019-02-18:
I have noticed that when I have a long print that will stay at a progress percentage for long time that if the bot is set to only notify every X seconds, it will notify multiple times for the same progress checkpoint if it is still at the same checkpoint percentage it already reported.

Example: I had a 68 hour print.  I have the bot set to only notify if at least 10 minutes have passed since the previous notification (so i'm not spammed with them on really short prints).  Given how long the print was, it usually sat at each percentage for 30+ minutes, so at each 10%, i got three to four notifications spaced approximately 10 minutes apart, as long as the progress was still at the same X0% status.

cameroncros, 2019-02-18:
That is interesting, sounds like Octoprint is notifying the plugin multiple times for a single percentage? As always, logs would help, but this might be an easy fix

tdub415, 2019-02-18:
I'll pull the log when I get off work; My best guess was the bot was polling for the progress when the timeout was over, and since it saw a X0% progress, it posted it again.

cameroncros, 2019-02-18:
Thats not how it works, progress events are pushed to the bot by OctoPrint. I'll just add a check that the current X% is not the same as the previous one.

cameroncros, 2019-02-20:
@tdub415 Are you using the periodic reporting function, or the percentage reporting function with the minimum timeout? If it is the first, I'd argue that the plugin is working as designed, but I can make it optional to hide duplicates if you want?

tdub415, 2019-02-20:
Percentage every 10% with 600 second timeout

tdub415, 2019-02-20:
Whoops, replied via email and not here.  I have it set to notify every 10% with a 600s (10 minute) timeout

closed 2019-02-21 by cameroncros

cameroncros, 2019-02-21:
Test v2.33 and let me know if it fixes the issue.

commit 71cc0c33 (2019-02-21): Fix #62 Prevent duplicate progress notifications