cameroncros/OctoPrint-DiscordRemote #63: Less verbose logging

megasaturnv, 2019-03-10:
Firstly, thanks for developing this great plugin.

I've been trying to reduce the logging of DiscordRemote to WARNING. Once applying the settings through the web interface, everything works fine and INFO logs from discordremote are not shown in octoprint.log or plugin_discordremote.log. However, after retarting the octoprint service INFO logs for DiscordRemote appear again.

After looking through the source code, I think it overrides the logging level setting here:
https://github.com/cameroncros/OctoPrint-DiscordRemote/blob/master/octoprint_discordremote/__init__.py#L146

OctoPrint: 1.3.10
OctoPrint-DiscordRemote: 2.34.0

Here is my logging.yaml file:
```
loggers:
  octoprint:
    level: WARNING
  octoprint.plugins:
    level: WARNING
  octoprint.plugins.discordremote:
    level: WARNING
  octoprint.util.connectivity_checker:
    level: WARNING
  octoprint.util.pip:
    level: WARNING
```

Logs (after octoprint service restart):
```
pi@raspberrypi-2 ~/.octoprint/logs $ date
Sun 10 Mar 17:10:03 GMT 2019
pi@raspberrypi-2 ~/.octoprint/logs $ tail octoprint.log
2019-03-10 17:06:50,631 - octoprint.plugins.discordremote - INFO - Heartbeat: {"d": 4, "op": 1}
2019-03-10 17:06:50,738 - octoprint.plugins.discordremote - INFO - Received HEARTBEAT_ACK message
2019-03-10 17:07:31,681 - octoprint.plugins.discordremote - INFO - Heartbeat: {"d": 4, "op": 1}
2019-03-10 17:07:31,801 - octoprint.plugins.discordremote - INFO - Received HEARTBEAT_ACK message
2019-03-10 17:08:12,732 - octoprint.plugins.discordremote - INFO - Heartbeat: {"d": 4, "op": 1}
2019-03-10 17:08:12,840 - octoprint.plugins.discordremote - INFO - Received HEARTBEAT_ACK message
2019-03-10 17:08:53,782 - octoprint.plugins.discordremote - INFO - Heartbeat: {"d": 4, "op": 1}
2019-03-10 17:08:53,894 - octoprint.plugins.discordremote - INFO - Received HEARTBEAT_ACK message
2019-03-10 17:09:34,831 - octoprint.plugins.discordremote - INFO - Heartbeat: {"d": 4, "op": 1}
2019-03-10 17:09:34,944 - octoprint.plugins.discordremote - INFO - Received HEARTBEAT_ACK message
pi@raspberrypi-2 ~/.octoprint/logs $ tail plugin_discordremote.log
2019-03-10 17:06:50,631 INFO Heartbeat: {"d": 4, "op": 1}
2019-03-10 17:06:50,738 INFO Received HEARTBEAT_ACK message
2019-03-10 17:07:31,681 INFO Heartbeat: {"d": 4, "op": 1}
2019-03-10 17:07:31,801 INFO Received HEARTBEAT_ACK message
2019-03-10 17:08:12,732 INFO Heartbeat: {"d": 4, "op": 1}
2019-03-10 17:08:12,840 INFO Received HEARTBEAT_ACK message
2019-03-10 17:08:53,782 INFO Heartbeat: {"d": 4, "op": 1}
2019-03-10 17:08:53,894 INFO Received HEARTBEAT_ACK message
2019-03-10 17:09:34,831 INFO Heartbeat: {"d": 4, "op": 1}
2019-03-10 17:09:34,944 INFO Received HEARTBEAT_ACK message
```

Not sure who's fault it is (probably mine :P) but thought I'd raise an issue and start here.

cameroncros, 2019-03-11:
I suspect your right. I think that line should be changed to lookup the logging level from the config. I'll get to it sometime this week

cameroncros, 2019-03-11:
Okay, you were right, just needed to get rid of the line

megasaturnv, 2019-03-11:
Okay, Thanks!

cameroncros, 2019-03-12:
Fixed in v2.35. Btw, thanks for the bug report detail, it is greatly appreciated.

closed 2019-03-12 by cameroncros

megasaturnv, 2019-03-12:
No problem, and thanks for fixing a fairly minor issue so quickly. I can now save a few writes on my pi's SD card without redirecting everything (including errors and warnings) to /dev/null. Cheers!

commit 07cb5a65 (2019-03-12): Fix #63 Logging is forced to DEBUG