cameroncros/OctoPrint-DiscordRemote #65: Notify for filament change

ferdaze, 2019-04-10:
Is there a feature to be able to send a notify to discode when the gcode pauses for a filament change using the M600?

cameroncros, 2019-04-10:
I am not familiar with how filament runout sensors work with Octoprint. I can investigate and let you know. I am fairly busy with other things at the moment, but will try get back to you in a week or so's time.

ferdaze, 2019-04-10:
I'm using Prusa MK3s and adding a M600 gcode which will cause the print to pause and ask for a color change. Users can have this site to add the gcode https://www.prusaprinters.org/color-print/ to it easy. 

The filament runout would also be a great notify but could be slit into a different request depending on what and how you sense the reason to send notifications.

cameroncros, 2019-04-11:
Ahh, okay, not filament runout, filament switchover. When the M600 is hit and the print pauses, do you get the "print paused" notification? I am not sure how "print paused" is triggered. I'll do some research, to be honest, I am not sure this is a feasable thing for discord remote to do, unless octoprint already exposes the event to plugins.

goscicki, 2019-04-11:
Yes, when M600 is hit "print paused" is triggered, as long as "//#define ACTION_ON_PAUSE "pause"" is uncommented in Marlin.

I thought this would be a good feature, but decided that if my print gets paused and I didn't do it, I'm going to go see what is going on anyway.

cameroncros, 2019-04-12:
So, if the "print paused" notification is getting sent, not much for me to change then? @ferdaze can you confirm that you receive the "print paused" notification?

cameroncros, 2019-04-26:
I'm going to close this ticket. If M600 causes a pause, that seems good enough. Also, I don't think its possible for plugins to detect GCODE codes, so I think its kinda out of my hands. If there is still a need for this, please re-open the ticket.

closed 2019-04-26 by cameroncros

ferdaze, 2019-04-26:
Thanks for trying. I do not see any notification when a M600 is sent from OctoPrint to the printer. I'm running a Prusa MK3s 3.7.0. I try some digging to see if there is something that can be sent to notify OctoPrint of a M600 and filament runout sensor.

goscicki, 2019-04-26:
I don't know what firmware you have, but if it is Marlin there is an option to enable this. It seems a little buggy though. When printing from an SD it causes a double pause for some reason.