cameroncros/OctoPrint-DiscordRemote #66: [Feature Request]

goscicki, 2019-04-11:
I think it would be helpful to be able to send gcode commands. For example /systemcommand gcode/M303.

cameroncros, 2019-04-11:
I have considered this before, but it is a very dangerous ability to open up to discord. A bad user could either accidentally or intentionally burn down your printer/house. Once the proper access controls are in, I'm happy to consider it, but I still feel a bit uncomfortable about it

goscicki, 2019-04-11:
Yeah, that makes sense. Couldn't your plugin simply filter gcode for safety? The only things that can be dangerous would be heating commands like "M104 S999". Even then the firmware should block anything above its safe limit. It's not like someone could flash dangerous firmware with gcode. You could make the plugin refuse anything above "M104 S250" or so. (also M109, M140, M190)

goscicki, 2019-04-11:
Or even just block heating commands if you have to.

cameroncros, 2019-04-11:
Even move commands could be dangerous to your printer, or anyone using the printer (repeatedly slam the motor into an axis until it breaks, move the bed or hot end rapidly while someone us pulling their print off, etc). I don't want to be responsible for maintaining the list of "unsafe" commands. (if I go through with it, I would be more inclined to make the gcode commands a whitelist the user has to fill in)

cameroncros, 2019-04-11:
Plunging into the object mid print is also a fire risk

goscicki, 2019-04-11:
I understand. I like the whitelist idea.

cameroncros, 2019-04-11:
Okay, well, i will see if I can implement this, but I am particularly busy atm, so it might take some time.

goscicki, 2019-04-11:
Ok, that's fine. I appreciate that you even responded. Thanks!

goscicki, 2019-04-21:
I have been testing this (DR-66) today and it seems great! My only suggestion would be to make the gcode non case sensitive. 
The same goes for file names in the /print command. It is a pain printing files with lots of uppercase from my phone.

cameroncros, 2019-04-21:
Thanks for testing, case insensitivity should be easy enough

goscicki, 2019-04-22:
I have been testing the case insensitivity today. It accepts lower-case gcode now, but the printer still only responds if it is upper-case. Perhaps it still needs to be converted to upper-case before sending it. I believe Marlin only accepts upper-case gcode. (The OctoPrint terminal converts it to upper-case.)

cameroncros, 2019-04-22:
That would make sense, I will convert it to upper before sending then

goscicki, 2019-04-23:
Awesome! It works great now. Thank you so much for all your hard work.

cameroncros, 2019-04-23:
No problem, thanks for your help testing it. I will release it in the next few days.

goscicki, 2019-04-28:
I just noticed this doesn't seem to work with "gcode system commands" plugin that allows OctoPrint to respond to custom gcode like OCTO0, OCTO1 etc. despite having them in the whitelist.

cameroncros, 2019-04-28:
Yep, according to the gcode specification, only G and M are valid codes. So I intentionally ignored other gcodes. I'll change it to be more open.

cameroncros, 2019-04-28:
Fixed. And released.

closed 2019-04-28 by cameroncros

goscicki, 2019-04-28:
Thank you so much!

goscicki, 2019-04-29:
This works great in the DR-66 branch. It doesn't seem to work on the master release though. Is it not merged?

cameroncros, 2019-04-29:
Yup, somehow didnt merge DR-66 in properly. Will fix and re-release