cameroncros/OctoPrint-DiscordRemote #74: permission denied after update

andypoo79, 2019-04-28:
Hi
i am unable to issue any commands to the printer after todays update. I have the * in the acxess box but still "permission denied", no change if i add my user ID it still responds "permission denied". is their any way to roll back?

many thanks

andypoo79, 2019-04-28:
please ignore my above post, i didnt realise i now have to give each user every command they can use. its a pain but i understand security and all

cameroncros, 2019-04-29:
@andypoo79 you should be able to put * in both boxes, and every user will have access to every command, does that not work for you?

Slayforfun, 2019-04-29:
I just discovered i have the same problem but I cant seem to find where I would input the asterisk for this, at least on on the plugin window of octoprint.

cameroncros, 2019-04-29:
[image]
You need to put * in both sides, and that allows all users to access all commands. If you want to lock it down, you can put a specific set of users, or a specific set of commands, or a combination of both.
If you need more information, please let me know.

Slayforfun, 2019-04-29:
Well the biggest problem, I think, is that I cant find that menu/header. The plugin configurations still look exactly the same as the way it did before I updated

cameroncros, 2019-04-29:
Ctrl-F5 to refresh the page?

Slayforfun, 2019-04-29:
yeah i just checked it on my laptop and there it did show, I feel stupid haha. Chrome hadnt refreshed the page as I opened the tab after a reboot of my pc. Thought it would be handled by my pi but shouldve been aware of this. Thanks for the patience, help and all your hard work :)

andypoo79, 2019-04-30:
> @andypoo79 you should be able to put * in both boxes, and every user will have access to every command, does that not work for you?

Hi yes this does work for me thanks

closed 2019-04-30 by andypoo79