cameroncros/OctoPrint-DiscordRemote #79: [Feature Request] gcode macros

goscicki, 2019-04-29:
The latest release seems flawless. Awesome work! I hope I'm not bugging you and asking too much. 

I think it would be very useful to anyone sending gcode, to be able to save a few simple macros. For example, the user names a macro "home" and sets it to "G28". They would type "/home" and it would send "G28". Or "/preheat" to send "M104 S60, M140 S200". 

Perhaps even allow variables like "/speed 80" could send "M220 S80", where "M220 S" is the macro and the trailing 80 is the variable.

I need to start learning python so I can help with this! I have some experience with C# but i'm self taught.

cameroncros, 2019-04-30:
If you want to take this on, I am happy to answer any questions and guide you through it if you want. It should be a fairly straightforward task I think.

goscicki, 2019-04-30:
Ok, thanks, I'll give it a shot. I love learning new things. I don't have much spare time, but I'll see what I can do.

goscicki, 2019-05-29:
Just letting you know, I still plan on working on this. I've just been VERY busy lately and haven't had the time.

cameroncros, 2019-05-29:
No problem

closed 2021-10-17 by cameroncros