cameroncros/OctoPrint-DiscordRemote #80: Regular messages improvement

EvilGremlin, 2019-05-08:
For small prints messages timings (every N seconds and every N%) often collide and just flood the chat, so my suggestion is this:
1. If difference in quantity for both types of messages estimated to be less than N%, prefer one type over the other.
2. Add a setting for a minimum interval between messages, with tolerance, like 30 minutes +-5%. If it's less - then skip message.

cameroncros, 2019-05-08:
You could just manually disable one message or the other? This just feels like a potentially very tricky solution that won't ever please everyone. If you want to work on it yourself, I am happy to review it and merge it, but I don't want to do this myself.

EvilGremlin, 2019-05-08:
Ehm... I'm feeling like an idiot. I just dug into sources and found that it's already can work the way i need. 
`# Skip if not a multiple of the given interval
            if int(data["progress"]) % int(tmp_config["step"]) != 0:
                return False`   
`# Otherwise work out if time since last message has passed.
            try:
                min_progress_time = timedelta(seconds=int(tmp_config["timeout"]))
                if self.last_progress_message is not None \
                        and self.last_progress_message > (datetime.now() - min_progress_time):
                    return False`
Basically, timeout works for percentage as well, if i didn't miss anything? So i can set it to report every 1% and 10 second, timeout to 3600 and get just 1 message a hour anyway?
Maybe note in settings that timeout applies to all status messages?

cameroncros, 2019-05-09:
I guess we are both idiots, I forgot about that :P But if that works for you, great. I am not sure where to document it, but I will leave this ticket open so that I remember to do it eventually (unless you want to do it)
- [ ] Document timeout behaviour.

commit 4a24b5aa (2019-05-29): Fix #80 Document minimum message timeout better.
closed 2019-06-01 by cameroncros