cameroncros/OctoPrint-DiscordRemote #81: Bot offline after octoprint 1.3.11 update

tdub415, 2019-05-15:
The bot appears to be offline since updating to 1.3.11

It's still reporting print progress per the settings, but the discord chat bot itself is offline and thus non-responsive to commands.

MikeSki3, 2019-05-19:
I kept seeing websocket error messages in the logs like this:
`websocket - ERROR - error from callback <bound method Discord.on_message of <octoprint_discordremote.discord.Discord instance at 0x6b13d148>>: on_message() takes exactly 3 arguments (2 given)`

So I tried updating the websocket creation with lambdas and that worked for me. The bot is back online.
`self.web_socket = websocket.WebSocketApp(socket_url,
                                                         on_message= lambda ws, msg: self.on_message(ws, msg),
                                                         on_error= lambda ws, msg: self.on_error(ws, msg),
                                                         on_close= lambda ws: self.on_close(ws))`

I'm not really sure what changed in the new version of Octoprint though.

cameroncros, 2019-05-19:
Im guessing websockets was updated. Thanks for your investigation @MikeSki3 , I will fix up the code tomorrow

closed 2019-05-22 by cameroncros