cameroncros/OctoPrint-DiscordRemote #83: I can't get mentions to work

Skullmonkey, 2019-06-09:
I have tried multiple formats to get mentions to work. I have tried @name and @name#(####) and the bot just prints them in plain text.

Here are screenshots of the settings I have tried: https://imgur.com/a/R2qzNIM

Am I missing something?

cameroncros, 2019-06-10:
i have no idea if that is the right way to do mentions. I will do some research and work it out

Skullmonkey, 2019-06-10:
If it isn't, you may want to update your documentation: https://github.com/cameroncros/OctoPrint-DiscordRemote#message-format

cameroncros, 2019-06-11:
That documentation is mostly from the older https://github.com/bchanudet/OctoPrint-Octorant#message-format project that this is forked from. I will work out what needs to be done and get back to you.

cameroncros, 2019-07-06:
According to: https://discordjs.guide/popular-topics/embeds.html#resending-embeds
"Mentions in embeds will not trigger a notification".

@Skullmonkey If you desperately need @mentions to work, please let me know and I will think of a way forward.

commit 15d66c93 (2019-07-06): Remove @mentions capability that doesnt actually work. Fix #83
closed 2019-07-06 by cameroncros