cameroncros/OctoPrint-DiscordRemote #84: Can't access files/timelapses from outside of home network

Sunlis, 2019-06-12:
It seems there used to be `/getfile` and `/gettimelapse` commands, which were removed in favour of download links in `/files` and `/timelapse` respectively.
My biggest usecase for DiscordRemote is interfacing with my printer when I'm not on my home network, since I don't want to open my octoprint instance up external requests.

I believe Discord has an 8mb upload limit, which would block some of the larger files out there. But would it be possible to add a command to attempt a file upload, and fail if the file is too large?

cameroncros, 2019-06-13:
This is certainly possible, however, most GCODE files and timelapses I have seen are well over the 8mb limit, which is why I gave up on that method.
I could look into sending the files in chunks, but I can't promise anything.
[ ] See if file uploading in chunks will work.

Sunlis, 2019-06-13:
Well I can't speak for everyone, but I use Octolapse to keep my timelapses short - guess it keeps them small too. The largest one I have right now is 6.8MB.

cameroncros, 2019-06-14:
https://stackoverflow.com/a/51296157 <- chunked zip, will work with 7z apparently

cameroncros, 2019-07-04:
haven't done timelapses yet, but added file upload for gcode. Will add timelapses soon

cameroncros, 2019-07-06:
@Sunlis I have added the /getfile and /gettimelapse functionality, if you are interested, you could test it by installing the feature/file_upload branch. Otherwise I will merge and release it in a few days

cameroncros, 2019-07-17:
@Sunlis the changes have been merged and released. If there are any issues, please create a ticket.

closed 2019-07-17 by cameroncros