cameroncros/OctoPrint-DiscordRemote #86: /listsystemcommands doesn't work

EvilGremlin, 2019-07-17:
It just doesn't respond to that command. Also, how do i run my custom commands from config.yaml ? It says syntax should be source/command, but what is source for my commands?

cameroncros, 2019-07-18:
It is working for me:

cameroncros, 2019-07-18:
https://github.com/foosel/OctoPrint/wiki/Cookbook:-System-Commands <- this is a good resource for how to add system commands. There is also a plugin to assist: https://plugins.octoprint.org/plugins/systemcommandeditor/

EvilGremlin, 2019-07-18:
Yeah, my config looks exactly like that. Doesn't work with bot. BTW, what distro you use? Mine is ARMBIAN 5.75 stable Ubuntu 18.04.2 LTS 4.19.20-sunxi. Maybe that's the case?

cameroncros, 2019-07-19:
It's not distro related, it was a bug in the parsing of the commands. Install the system commands plugins, and try again.

EvilGremlin, 2019-07-19:
Thanks man, will try this evening.

EvilGremlin, 2019-07-19:
Okay, i got something new in log now:
2019-07-19 21:25:06,453 - websocket - ERROR - error from callback <bound method Discord.on_message of <octoprint_discordremote.discord.Discord instance at 0xa9b7bb98>>: 'ascii' codec can't encode characters in position 0-8: ordinal not in range(128)
Well, i changed names to latin and finally got commands list, but ASCII? WTF? It's 2019, UTF-8 is default standard everywhere now.

cameroncros, 2019-07-20:
Probably mine, Gina is German, I suspect she has proper utf-8 support. I will check that later and push a fix if it's my end.

cameroncros, 2019-07-20:
So, definitely my end. Or python 2's end, but I believe it is fixed. If there are any issues, let me know.

closed 2019-07-20 by cameroncros

EvilGremlin, 2019-07-20:
Yeah, now it's perfect. I guess it's more python's heritage than bug.

cameroncros, 2019-07-21:
Yeah, I learnt something as well. At least in python 3 you can use str, as it is unicode, but until octoprint jumps up to python 3, it's out of my hands :(

commit 6c30b1e4 (2019-07-19): Fix #86 /listsystemcommands fails if the name is not set
commit ff5b8971 (2019-07-20): Convert str to unicode to properly support unicode. Fix #86