cameroncros/OctoPrint-DiscordRemote #88: [Feature Request] "Mask External IP" or Subdomain support.

CyberSecGuy, 2019-07-25:
Similar to #27 I'd like to be able to submit a static 'mask' for the external ip - I still want it to show up, but I've setup an A Record on my Domain and a VHOST Reverse Proxy to connect to Octopi on my network externally. Maybe this can be an alternative option to "Hide External IP"

Concept Design: https://i.imgur.com/33bwMif.png

I've done this manually editing your scripts (but need to re-edit each update.) Here's what I have on my discord: https://i.imgur.com/fNjyFjh.png

cameroncros, 2019-07-26:
Sounds find to me, if you create a PR I would be happy to merge it?

CyberSecGuy, 2019-07-26:
I now realize how what I said implies I already did the code. That is my mistake, I simply edited the html in Chrome Inspect Element.

Having said that: I would love to try it out myself, I have limited knowlege of python. I will (hopefully) return with a Merge request.

cameroncros, 2019-07-26:
I am happy to leave it to you, if you need any help, let me know.

shaynemk, 2020-04-24:
@CyberSecGuy Not sure if you made any progress on this, but I drafted it up real quick for the basic functionality. There are a few more sanity checks on my list of things to do, but I'll get to those later.

closed 2020-05-10 by cameroncros