cameroncros/OctoPrint-DiscordRemote #90: {Feature Request] Show the current prefix under "playing"

OctopusProteins, 2019-07-31:
Hello,

I would like to see a feature similar to this added:
[image]
to display the current set prefix and/or help command.

I am willing to work on adding this feature, just thought I'd make a feature request to keep track of it. Thoughts?

Thanks,
OP

cameroncros, 2019-07-31:
Sounds good to me, not to scope creep it or anything, but the status would be cool place to show the printers current status as well. Eg: `Idle. /help to list commands` or `Printing file.gcode - 35%. /help to list commands`

But, do what you want, I'll merge it if it is good.

OctopusProteins, 2019-08-09:
So I was thinking about how to approach this, and I think a rotating status message would work well. So rotating between /help, and the current status (printing 35%, paused, offline, etc.). In order to rotate the status text, a background task is needed. I'm not especially familiar with python, so I'm having some trouble using asyncio with something like this. https://stackoverflow.com/questions/46267705/making-a-discord-bot-change-playing-status-every-10-seconds
What's the best way to approach this?

cameroncros, 2019-08-09:
Best approach would be to create a seperate thread to periodically send updates I think. Look at the heartbeat thread in the discord file.

OctopusProteins, 2019-08-09:
I'm having some trouble finding the `discord` class. For example to use `discord.Game` or to access the `discord.Client()` object. Where can I get references to these?

cameroncros, 2019-08-10:
The discord library used by this plugin was made by me for this plugin. The source is the discord.py file. The other discord libs are all python3 specific, which didn't work for octoprint. So, your gonna have to extend the class if you want game features.

API documentation.
https://discordapp.com/developers/docs/resources/channel

SgtKiLLx, 2020-10-07:
Wonder if any progress has been made , or if this has been abandoned.

cameroncros, 2020-10-08:
Afaik, this has not progressed. You are welcome to take up the mantle though, and I will merge it

OctopusProteins, 2020-10-08:
Correct, haven't made any actual changes yet. I have a plan of action, but just have not had a chance to implement it yet. Might get to it after exams in a month or two. If you're interested in picking it up and have any questions, feel free to ask!

SgtKiLLx, 2020-10-08:
Ahh, thanks guys for the reply. I'm not skilled with python or plugin dev. I did find someone in the octoprint discord, that might get to it, but not anytime soon. I was just thinking of this, posted , then seen this post. Love the idea. If yous have discord, let me know. I'd love to even watch from the sidelines. Glad I got it going again after a year. Bedtime for me tho.

cameroncros, 2020-10-08:
@SgtKiLLx I am more than happy to help guide you through making a change if you are interested in learning. I was not skilled at Python or Octoprint dev until I started this project.

SgtKiLLx, 2020-10-08:
> @SgtKiLLx I am more than happy to help guide you through making a change if you are interested in learning. I was not skilled at Python or Octoprint dev until I started this project.

Awesome. Can I add you on discord?

cameroncros, 2020-10-08:
https://discord.gg/nn8dYN <- this is the discord channel I use for testing/development, this is probably the easiest way forward

SgtKiLLx, 2020-10-09:
> Correct, haven't made any actual changes yet. I have a plan of action, but just have not had a chance to implement it yet. Might get to it after exams in a month or two. If you're interested in picking it up and have any questions, feel free to ask!

Hey. You needa join the discord.

closed 2020-10-18 by cameroncros