cameroncros/OctoPrint-DiscordRemote #91: Temperature reading from Enclosure Plugin

yamanub, 2019-08-06:
This is not really an issue. I am using the Enclosure plugin and currently have a temperature sensor connected through it. I was wondering if there is a way to display the current temperature reading from this sensor in discord?

cameroncros, 2019-08-07:
Can octoprint show the temperature sensor already?

yamanub, 2019-08-07:
It shows it on the Enclosure plugin tab only (or in the navbar if the option is selected in the plugin)

cameroncros, 2019-08-07:
Okay, well, without looking too deeply into it, the enclosure command plugin could be extended to give you that information? I dunno if that is within your abilities, but you could try that. I am more than willing to guide and assist if need be, but I don't have much spare time to look into it myself at the moment.

yamanub, 2019-08-07:
My abilities are pretty weak in that aspect. I might have a friend who can help. Let me see what I can come up with. I would assume the /outputstatus results is where the temp could show since that command is getting feedback from the enclosure plugin outputs? I really don't know what I am talking about... lol

Let me see if my friend can help and I will keep you posted. Thanks

cameroncros, 2019-08-07:
I think your pretty close, but I don't know how the enclosure plugin works for temperature sensors.

Have a look at https://github.com/cameroncros/OctoPrint-DiscordRemote/blob/master/octoprint_discordremote/command_plugins/enclosure_control.py, 

You'll probably need to add a /outputtemp command, and hook it up to the correct enclosure plugin API. Good Luck

OctopusProteins, 2019-08-09:
hey @yamanub, i created the enclosure plugin originally, if you still need some guidance. 

As cameroncros said, check out enclosure_control.py. There, you can add new commands. 
The API for the enclosure plugin is here:
https://github.com/vitormhenrique/OctoPrint-Enclosure/wiki/API-Control
All the data you need access to should be accessible through the API.

Let me know if you have any questions. I also had little idea what I was doing when I started the plugin, so I'm happy to explain.

closed 2021-10-17 by cameroncros