cameroncros/OctoPrint-DiscordRemote #92: Discord snapshots continue if the printer disconnects

jgrills, 2019-08-14:
My heating element is having issues and sometimes won't heat up. When the printer disconnects, the discordremote plugin doesn't realize the printing is effectively stopped and continues to spam discord with photos when the periodic timer expires. I do realize if I was using the percentage with a minimum timer it would probably have solved the issue as well (since the print isn't progressing with the printer disconnected). I'll switch over to that, but still wanted to report the bug.

Perhaps an `if (isConnected())` check before sending a snapshot...

cameroncros, 2019-08-14:
Interesting, I can't say im in a hurry to fix that if im honest, but I will look into it at some point

cameroncros, 2019-08-17:
Think i have fixed this. @jgrills are you able to manually test it yourself?

closed 2019-08-17 by cameroncros

jgrills, 2019-08-19:
I'll try, though I just replaced my heating element and think I've addressed the issue. I think if I unplug the temperature sensor and try to start a print I suspect it will happen.

jgrills, 2019-08-19:
Appears fixed! Thanks, I'll go back to using periodic updates.

commit cd852b41 (2019-08-17): Fix #92 Check if the printer is still printing when doing a periodic
print