cameroncros/OctoPrint-DiscordRemote #93: Remove "Extruder Temp (chamber)"

andrelec1, 2019-08-16:
I have a Prusa MK3 and dont have this value...

when i /status i see : 
[image]

Juste provide a checkbox like external/local ip for showing this value :)

cameroncros, 2019-08-16:
If you create a PR, i would be happy to merge that. But, I am not particularly concerned about that, given it still functions.

andrelec1, 2019-08-17:
Sorry, i'm not python dev :/ 

I think this is in fault :
https://github.com/cameroncros/OctoPrint-DiscordRemote/blob/cdb91cc0597cc1da0872d2c28fc38c6c5841d861/octoprint_discordremote/command.py#L312-L315

but when i go throuth http://octoprint/api/printer
the json look like :
```json
{
  "state": {
    "flags": {
      "cancelling": false, 
      "closedOrError": false, 
      "error": false, 
      "finishing": false, 
      "operational": true, 
      "paused": false, 
      "pausing": false, 
      "printing": true, 
      "ready": false, 
      "resuming": false, 
      "sdReady": false
    }, 
    "text": "Printing"
  }, 
  "temperature": {
    "bed": {
      "actual": 50.9, 
      "offset": 0, 
      "target": 50.0
    }, 
    "tool0": {
      "actual": 215.8, 
      "offset": 0, 
      "target": 215.0
    }
  }
}
```

so didn't understand where the "chamber" key come from ...
i try to search in all project ... didn't found anything :/

commit 785fb36e (2019-08-18): Fix #93 Hide tempretures that dont have values.
closed 2019-08-18 by cameroncros