cameroncros/OctoPrint-DiscordRemote #94: Add layer progess ...

andrelec1, 2019-08-16:
In /status can you add the current layer / total layers  ...

cameroncros, 2019-08-16:
Maybe. I vaguely recall attempting this, and finding that there is no way for a plugin to get the current layer. I may look into this in the future, but if you submit a PR, i would be happy to merge it.

andrelec1, 2019-08-17:
I'm not python dev, but actualy a use this plugin :
https://plugins.octoprint.org/plugins/layerdisplay/

and you can get json info throught; http://octoprint/api/plugin/layerdisplay
```json
{
  "layerString": "57 / 688"
}
```

cameroncros, 2019-08-18:
I'm hesitant to rely on other plugins for core functionality. Im not going to close this ticket, so if someone comes along, they can implement it, but I will likely not do it.

andrelec1, 2019-08-22:
I didn't understand why the layer progression isn't add natively in Octoprint ...

cameroncros, 2019-08-23:
Me neither to be honest. I guess it is hard to work it out on vase-mode prints?

andrelec1, 2019-08-23:
Vase-mode is juste a print with lot of layer ...
All plugin that track the layer progression analyse the gcode for counting `G1 Zxxx` instruction

closed 2021-10-17 by cameroncros