cameroncros/OctoPrint-DiscordRemote #96: Support Octolapse

andrelec1, 2019-08-16:
I think most of people use octolapse instead of build-in timelapse feature ... 

It would be anwsome if i can sync the shot by octolaps and sending picture in discord ...

Octolaps can call external script ...  i think the most simple way its juste provite a api/url for sending message when octolaps take the picture ... 

(  https://plugins.octoprint.org/plugins/octolapse/ )

cameroncros, 2019-08-16:
Interesting idea, not sure how acheivable. If you are willing to submit a PR, i would be happy to review it.

cameroncros, 2019-09-12:
@andrelec1 Okay, i have done some thinking about this, and it sounds like what you actually want it an API endpoint that allows any old plugin to send a message to discord?
If so, that might be extremely easy to implement, and I will start doing that now. Will you be able to test it once its implemented?

andrelec1, 2019-09-13:
Yes, just a api ;) 
And yes, i can test and help you to debug ...
( but not right now, let me time to buy another RPI for my printer ^^ https://streamable.com/71ovu )

cameroncros, 2019-09-14:
@andrelec1 I have added an API endpoint to allow sending a message. Please reopen this ticket if there are any issues with it. https://github.com/cameroncros/OctoPrint-DiscordRemote#api

closed 2019-09-14 by cameroncros