cameroncros/OctoPrint-DiscordRemote #97: object of type 'float' has no len()

stevegood, 2019-08-20:
When invoking `/status` I am seeing the following error in octoprint.log

```2019-08-19 17:04:53,337 - websocket - ERROR - error from callback <bound method Discord.on_message of <octoprint_discordremote.discord.Discord instance at 0x6e8ef8c8>>: object of type 'float' has no len()```

Discord Remote version: 2.45.0
OctoPrint version: 1.3.11

cameroncros, 2019-08-20:
Sounds like a bug, i will investigate and try fix it soon

stevegood, 2019-08-20:
Just guessing but I presume the problem is here: https://github.com/cameroncros/OctoPrint-DiscordRemote/commit/785fb36e73a205c4061c5348c9945675bb34e290#diff-efc848f6e4f3d95c93e0703d800605dcR315

cameroncros, 2019-08-20:
> Just guessing but I presume the problem is here: 785fb36
> #diff-efc848f6e4f3d95c93e0703d800605dcR315

Yup, the unit tests were failing as well, but i didnt notice :/
Should be fixed now.

commit cf837890 (2019-08-20): Fix #97. Accidentally treated float as a string.
closed 2019-08-20 by cameroncros