cameroncros/OctoPrint-DiscordRemote #98: Put /help in permission system

andrelec1, 2019-08-20:
I have some stupid guy's in my discord that spam /help , and try every command -_-" ....
Can you just put the /help in the restriction systeme like other command ?

cameroncros, 2019-08-21:
My rationale for leaving help outside the permissions system is that I wanted it to be always available.
Moving it into the permissions system does somewhat compromise that, but I am happy to do it. Note that if it becomes a problem later down the line, i might revert it.

Also, I strongly suggest you use discords permissions to block users from posting to your discord channel. If I made a mistake in the permissions system, they could very easily damage your printer or start a fire. Discords permission system is far better tested than the poor one I put together.

andrelec1, 2019-08-22:
you absolutely right...

commit 206a842f (2019-08-31): Fix #98. Move help into the permissions system.
closed 2019-08-31 by cameroncros