cameroncros/OctoPrint-DiscordRemote #99: /status command seems to be not working

tdub415, 2019-08-21:
I have noticed recently that /status doesn't seem to be working.  Other commands like /snapshot are still working for me.

cameroncros, 2019-08-22:
Can you find any log messages that seem relevant?

tdub415, 2019-08-22:
2019-08-21 23:44:47,112 - websocket - ERROR - error from callback <bound method Discord.on_message of <octoprint_discordremote.discord.Discord instance at 0x73672738>>: object of type 'float' has no len()

tdub415, 2019-08-22:
it looks like this may have been fixed in v2.46, which i was almost positive i had updated to, but it looks like I'm still on 2.45

cameroncros, 2019-08-22:
Yeah, sounds like the issue fixed in 2.46, so I think i will close this. Feel free to reopen it if you have further issues.

closed 2019-08-22 by cameroncros

tdub415, 2019-08-22:
Yea, 2.46 fixed it, but looks like 2.46 is installing labeled as 2.45.  I
updated the comments of
https://github.com/cameroncros/OctoPrint-DiscordRemote/issues/87 since it's
doing it again.