cameroncros/OctoPrint-DiscordRemote #100: [Feature request] Support temperature viewing commands, triggers, and macros

Moeser, 2019-08-23:
Temperature of the print head and print bed are very important.  OctoPrint keeps track of these values, but DiscordRemote doesn't use them in any way.  This is a feature request to add full support for reading the temperature.  (please *do not* add support for writing or changing the temperature, since no one wants a goofball on discord heating their print bed up to match the surface of the sun)

Commands:
Being able to use `/temp` to get all temperatures, or `/temp bed` or `/temp tool` for specific temperatures would be nice.  Bonus points if you can display the temperature graph gif that OctoPrint renders.

Macros:
Being able to add macros to the messages that resolve to a temperature would be very handy.  Can you imagine a line in discord like: "The print is 50% complete! The print head temperature is 204c, and the print bed temperature is 55c!"

Triggers:
Similar to the above, it would be nice to add a triggered message if the temperature goes above or below a specific threshold during the print.  For example: "The print bed temperature has gone above 110c and is now 112c!  Someone check on this thing before it burns itself up!"

cameroncros, 2019-08-24:
/status already gives you the value of all temperature sensors.

The macro idea is reasonably straight forward, anyone should be able to make this work, if I find time I'll do it.

The triggers idea is a lot more involved. DiscordRemote does not regularly monitor the printer, and that kind of task would ideally be left to Octoprint itself (or your printer firmware). I am very hesitant to implement such a feature, as it is way out of the scope of DiscordRemote. If you wish to do it anyway and submit a PR, I will consider it, but I suggest you look for another alternative. I am willing to discuss it further if you wish.

cameroncros, 2019-08-31:
> On /status giving temperature sensor values, that's true, and works fine. Having the OctoPrint temperature graph would still be nice, but I'll concede that it might be overkill.

Getting the graph would be awesome, but would require pulling in some kind of library to generate the graph. The octoprint one is generated by the browser.
 
> On the triggers idea, I'm not sure I follow the reasoning that notifying Discord about out-of-limit temperature readings should be in the firmware or in OctoPrint. I don't think either OctoPrint, or printer firmware would know how to send messages to Discord natively. I guess it could be part of a different Discord plugin, but then I'd have two Discord plugins to authorize and maintain, which seems... weird.

Octoprint definitely cannot send messages to discord, however, monitoring the printer is what Octoprint is designed to do. Ideally, i would prefer for Octoprint to monitor the temperature, and when it goes above/below a set limit, it would generate an event that DiscordRemote could then report.

> I get your point that DiscordRemote doesn't currently monitor the printer regularly. I thought maybe there was a regular polling thread or something. Do the events which trigger print progress updates have temperature info? If they do, maybe saving the temperature states at that time would work instead of regular polling?

The temperature can be gotten at any time, and theoretically, DiscordRemote could create a thread to poll the printer regularly. It just feels wrong to do that. Also, if your goal is to get the temperature value as soon as it goes above/below, waiting for the next progress report might be too late, and the printer may already be on fire. I think Octoprint is the right place to do that kind of monitoring.

cameroncros, 2019-09-12:
@Moeser I am going to close this ticket as out of scope. If OctoPrint ever adds temperature monitoring with an event for over/under temperature, i will implement the event. If you want to see this, i suggest you make a feature request on the OctoPrint issue tracker.

closed 2019-09-12 by cameroncros