cameroncros/OctoPrint-DiscordRemote #102: [Request] Do something before and after image capture

lloyb, 2019-09-01:
Hello, Sorry if this is not the place to put in a request, I signed up to be able to post this.
My printer sits in a dark room but I have a lamp next to it with a Kesa plug. When i remote connect in octoprint, I trigger my light to see how things are going then turn it back off.
I love this discord remote but the images it sends me are black unless its daytime.
Would it be possible to have the script trigger the light, then take the picture, then turn it off?

I'm guessing I could setup the light with a gcode trigger so the setting would be, 
gcode before photo: 
delay: (to count for how long it takes for the light to turn on)

then after image
gcode:

Or is this something it can already do but I'm missing the settings page?

Thanks for making the remote, I hope I'm not asking too much

cameroncros, 2019-09-01:
As it happens, there is such a feature, but it is a very untested and underused feature.

I am not 100% sure how to enable it, but you can run a script before and after every event.

You have to manually edit your octoprint config, and in the discord remote section add a "allow_scripts: True" field. Note that the config is yaml, which is fussy about indentation and does not allow tabs. I can give more guidance if you need it.

Once you have that set, you can open the settings page for DiscordRemote, and there should be a before and after script section. Add a script in the before, to trigger the light, and one after to turn it off.

The scripts can be any kind of executable script.

Please let me know how it goes, I have never used this feature so more information and documentation would be greatly appreciated.

cameroncros, 2019-09-07:
@lloyb I'll close this issue now, but if you have any issues, please reopen it and I will try help.

closed 2019-09-07 by cameroncros